ptx2/gymnasticon #3: Blocking problems installing to Raspbian VM Guest

gojomo, 2020-08-03:
I don't have a Pi handy, so figured I'd give it a try in a Raspbian VM (under VMWare Fusion), using image from https://www.osboxes.org/raspbian/#raspbian-2019-09-30-vmware - hey, it might work, right?

Not an `npm`/`node` pro, but after dealing with various other version/library issues, installing from npm (`npm install -g gymnasticon`) gave various warnings but eventually wound up with an executable. However, any execution gave following stack:

```
root@osboxes:~# gymnasticon 
internal/modules/cjs/loader.js:638
    throw err;
    ^

Error: Cannot find module '../bikes'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
    at Function.Module._load (internal/modules/cjs/loader.js:562:25)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/usr/local/lib/node_modules/gymnasticon/lib/app/app.js:20:14)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
```

Hmm, OK. `bikes` is obviously in git project, what if I try cloning git dir and using the `npm link` installation?

Although much installs/compiles, it eventually reaches blocking error where a `gymnasticon` executable is not even established, specifically ending in (this example is from a second `npm link` execution in the same dir):

```
root@osboxes:~/gymnasticon# npm link
npm WARN gymnasticon@1.0.0 No repository field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connect@2.0.0 (node_modules/xpc-connect):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connect@2.0.0: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"ia32"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.12 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.12: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"ia32"})

audited 655 packages in 3.613s

43 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities

npm ERR! code ENOENT
npm ERR! syscall chmod
npm ERR! path /usr/local/lib/node_modules/gymnasticon/lib/app/cli.js
npm ERR! errno -2
npm ERR! enoent ENOENT: no such file or directory, chmod '/usr/local/lib/node_modules/gymnasticon/lib/app/cli.js'
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent 

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-08-03T19_03_35_028Z-debug.log
```
If either of these are clear indicators of a missed step or easy fix, please let me know!

(Perhaps some of those 'OPTIONAL' dependencies aren't really optional?)

Happy to provide longer logs of the attempts if they'd be helpful.

Seeing more-clearly stated versions (for `npm`, `node`, etc) for the place where you know this works might also help, in case the sweet spot is between the older Raspbian defaults and the `latest` I've also tried. 

Thanks, in any case, for the big progress in re-enabling these bikes' functionality! (You may also want to coordinate with the author of the [MacOS Derailleur](https://github.com/benwithjamin/Derailleur) project, @benwithjamin, to share protocol discoveries & make functionality available on other platforms. Many people probably have a desktop/laptop/spare-handheld in same room as their Flywheel that could also do the protocol-translation, though of course a dedicated Pi that just plugs into the bike & "does the right thing" is pretty slick.)

ptx2, 2020-08-03:
Hi, thanks for reporting this.

I have released 1.0.1 to npm which I believe should fix that issue. Some necessary files weren't included in the 1.0.0 package.

```
npm install -g gymnasticon@1.0.1
gymnasticon --version
```

Re: multi-platform. I have only tested on Raspbian but it should work on x86 Linux and macOS. The main thing is having a Bluetooth 4.1+ adapter that can do multi-role. I have tested somewhat on macOS but lack a Bluetooth 4.1 adapter on that machine so can't test fully. I opened a new issue for macOS #4

gojomo, 2020-08-04:
Thanks for quick fix! Installing 1.0.1 overcame previous error. There's a new error, but it may not have a software fix. Specifically:

```
root@osboxes:~# gymnasticon
noble warning: adapter does not support Bluetooth Low Energy (BLE, Bluetooth Smart).
               Try to run with environment variable:
               [sudo] NOBLE_HCI_DEVICE_ID=x node ...
[2020-08-04T04:13:02.745Z] Error: Bluetooth adapter state: unsupported
    at App.run (/usr/local/lib/node_modules/gymnasticon/lib/app/app.js:116:40)
    at process._tickCallback (internal/process/next_tick.js:68:7)
```

Issue is probably the somewhat limited Bluetooth sharing supported VMWare Fusion. (Seems like: paired devices can be shared, but not all BLE in/out.) I should have a USB Bluetooth dongle in a couple days that may server as a work-around (as an entire USB device *can* be shared to the guest VM, so it'd be able to do everything the dongle & its driver can do.

But, this issue can be considered resolved, thanks.

gojomo, 2020-08-06:
OK, new related issue. Using an external BT dongle, which  successfully appears in the `lsusb` output & changes the `hci0` device BD address when it's inserted (which makes me think it's displaced the default limited BT sharing). But, there's now a more run-of-the-mill JS error stack:

```
root@osboxes:~# gymnasticon
[2020-08-06T04:18:44.411Z] connecting to bike...
[2020-08-06T04:18:44.414Z] TypeError: (0 , _events.on) is not a function
    at scan (/usr/local/lib/node_modules/gymnasticon/lib/util/ble-scan.js:24:32)
    at FlywheelBikeClient.connect (/usr/local/lib/node_modules/gymnasticon/lib/bikes/flywheel.js:71:48)
    at App.run (/usr/local/lib/node_modules/gymnasticon/lib/app/app.js:119:23)
```

Any thoughts?

ptx2, 2020-08-08:
It looks like the Node.js version might be too old. If it's installed from the Raspbian Buster apt repositories then it's likely the case as that's 10.x. The latest 12.x (12.8.3) release will work.

For armv7l (Raspberry Pi 4) there's a binary build here:
https://nodejs.org/dist/latest-v12.x/

For armv6l (Raspberry Pi Zero W) there's a binary build here:
https://unofficial-builds.nodejs.org/download/release/v12.18.3/

This is not related, since you're running as root, but another thing to be aware of is if you end up running as an unprivileged user is to set the Bluetooth capabilities somehow (e.g. with `setcap` on the node binary, or with `AmbientCapabilities=` as in the example systemd unit file).

I've updated the README to include all of the above info. Good luck, let me know if you have any luck or problems!

ptx2, 2020-08-09:
Not sure if you have access to a mac but I got it working on macOS 10.13, 10.14 (possibly 10.15+ but don't have a machine to test) see #4 for details

gojomo, 2020-08-20:
So, some updates that are off the original issue, but reflective of my latest progress & blocks:

I'd thought running a little Raspbian-like VM inside my MacOS would be a good way to test things before tackling a real Raspberry PI, but that was a disaster several times over. 

First, the Rapbian VM image from osboxes.org totally resisted any attempts to get a recent NodeJS installed. (A [question on SO logs some of the frustrations](https://stackoverflow.com/questions/63321424/how-to-install-working-recent-node-stable-12-x-on-raspbian-debian).) Just doing a vanilla Debian install has worked much, much better. 

Second, VMWare Fusion VMs don't get full Bluetooth LE access - just a sharing of paired devices. OK, I thought, USB Bluetooth dongles are $6-10, VMWare allows raw USB sharing, I can give my VM full Bluetooth LE that way. That works, in Linux! Woohoo! But, due to a [festering Apple bug in recent Mac/MacOS versions](https://discussions.apple.com/thread/250944058), **plugging in an external Bluetooth dongle persistently disables onboard Bluetooth, perhaps permanently**. So a <$10 dongle may eventually send me to Apple for a motherboard replacement. But, using the dongle still works – well in Linux, a bit flaky in MacOS – in the meantime. 

So, I can run Gymnasticon! And it definitely finds my Flywheel. But:

(1) If the Flywheel had been connected to anything else, or had recently 'spun down' from a period of activity, there tends to be an exception in Gymnasticon like the one shown here: 

[image]

I suspect that specifically upon ending/timing-out a BLE session there's some message sent that's not of the expected form. 

(2) While I can easily restart after such an error, and while pedaling see accurate readouts like at the top of that screenshot, I don't see any evidence, yet, of the software rebroadcasting activity data in another format. I don't see any clearly-identified 'Gymnasticon' BLE sources in a BLE Scanner app (the same one that sees "Flywheel" clearly, even while attached to Gymnasticon). I also tried using the 'Rouvy' app's `Training & sensors` -> `Select trainer` -> `Other` -> `General BLE FTMS trainer (beta)` option - no dice, even as the Gymnasticon log is showing cadence/power/etc log lines. So, I'm wondering if there's some extra step or module I've missed for rebroadcasting. 

(I'm reluctant to install NodeJS & a Node app on my main MacOS machine as opposed to a silo'd VM, and the dongle thing has me even less adventurous than usual.)

But, in other frontiers: have you seen the experimental-in-Chrome (& maybe elsewhere) "Web Bluetooth API" – https://googlechrome.github.io/samples/web-bluetooth/ ? Plausibly, in Chrome on MacOS, Android, & elsewhere, it could let a **web page** detect & read BLE devices, & perhaps send new BLE data, as well.

ptx2, 2020-08-25:
Glad you got the client part working in the VM!

As for the server part not working (i.e. it not showing up in Zwift), the first thing that comes to mind is whether the dongle supports BLE multi-role (simultaneous client and server). Do you have any details you can share on the dongle?

Some things to try:

1) Run just the server part of Gymnasticon (no client connection to the Flywheel bike) and see if it shows up in Zwift. Something like `gymnasticon --bike bot --bot-cadence 60 --bot-power 100`. If this works, it suggests to me the issue is with multi-role. If it doesn't work, I would say to check permissions/capabilities although it looks like you're running as root so I imagine that isn't it.
2) Run gymnasticon normally and in another shell run btmon to see if there are any failures regarding advertising `btmon -w trace.btsnoop`.

There is an experimental macOS version in #4 that does seem to work, but I know you said you'd rather not run it on your mac.

Perhaps of more interest: I recently released a [custom Gymnasticon SD card image](https://github.com/ptx2/gymnasticon#quick-start-install-gymnasticon-sd-card-image) that you can just download and boot without having to install or configure anything.

And, yes! I actually made a proof-of-concept using WebBluetooth a little while ago that connects to the bike and shows the stats in the browser. It was very easy to get going and worked fine. There was no server/peripheral API in WebBluetooth when I looked though which would be needed to fully implement something like Gymnasticon. I think this could be a cool way to distribute a cross-platform calibration tool for the Flywheel bike.

Thanks for reporting that unhandled exception issue. I've opened #10 to track that.

ptx2, 2020-10-10:
Hey, I'm going to close this one out as it's sat inactive for a while now. Feel free to reopen.

closed 2020-10-10 by ptx2