ptx2/gymnasticon #4: Test macOS

ptx2, 2020-08-03:
This should be possible as the underlying Bluetooth libraries (noble and bleno) both support macOS.

I have tested the client (to/from bike) and server (to/from zwift) components individually and they do work on macOS 10.13.
However it is not possible to fully test the app end-to-end on this machine as its Bluetooth adapter does not support multi-role (client and server at the same time). macOS 10.13 is also quite old.

Would be good to test on newer macOS with a Bluetooth 4.1+ adapter that supports multi-role.

ptx2, 2020-08-09:
There is a working experimental version for macOS 10.13 and 10.14 here:

https://github.com/ptx2/gymnasticon/tree/macos

It may work on more recent macOS but don't have a machine to test.

ptx2, 2020-08-09:
To test:

    git clone https://github.com/ptx2/gymnasticon
    cd gymnasticon
    git checkout --track origin/macos
    npm install
    npm run build
    npm link
    gymnasticon --help

jeremydk, 2020-09-06:
Working from 10.15 -- Only did a one-legged test (bot profile to Garmin), but successfully seeing power/cadence data flow.  It does appear that the Garmin wants more frequent cadence updates -- 100 bot cadence is showing up as flashing --- and 104, but I'm not sure that's related to the underlying OS at all.

javimontero, 2020-09-23:
> Working from 10.15 -- Only did a one-legged test (bot profile to Garmin), but successfully seeing power/cadence data flow. It does appear that the Garmin wants more frequent cadence updates -- 100 [...]

@jeremydk To Garmin? Can you share details? Are you sending cadence/power to garmin computer/watch?

jeremydk, 2020-09-23:
Yes.

In order for Gymnasticon to be seen as a valid power meter on my Edge 530,
I needed to apply to path from this branch to remove the descriptors in the
characteristic. While I was doing development, I was using a MacBook Pro
running 10.15 between my Peloton and the Garmin.  My “production” setup is
now a raspberry pi zero w, but the Garmin behavior remains the same.

vodlogic, 2020-10-18:
I've been able to successfully broadcast Peloton measurements to an Zwift running on an iPad using my MBP using this branch, so wondered if there was anything else left preventing this from being merged?

barkoneus, 2020-12-13:
Got this to work with a couple modifications:

1) Got the following error during the npm install step:

npm ERR! Command failed: git checkout b7acf8b72f19056e81f838ab611005b21f997777

Got around it by replacing the reference with a valid checkin reference to bleno (63f5ba92c62cab33580f93b010173bd9b1ec9221) in package.json and package-lock.json

2) Needed to to run npm link as sudo

Machine:
macOS 11.0.1 / M1 MBP

That said I can't actually confirm if the program is working correctly, as it just says "connecting to bike..." but doesn't do anything else (trying to use with a Schwinn IC4 so probably need to modify some config files or something)

ptx2, 2021-01-23:
Thanks for trying this out. I had actually merged macOS support into the main branch a while ago but forgot to close this ticket.

@barkoneus There isn't currently support for Schwinn IC4 in Gymnasticon. If it can produce power and cadence information it should be easy to add support though. However looking at the IC4 product page it claims to work with Zwift out-of-the-box so perhaps Gymnasticon isn't needed? Feel free to open a new issue for IC4 support if you want!

closed 2021-01-23 by ptx2