ptx2/gymnasticon #5: Typo inside Readme

rmaster78, 2020-08-14:
I was able to get it all working on my raspberry pi zero but ran into some install issues (perhaps this is due to me not knowing anything about programing or Linux etc, but was able to figure it out by searching the error messages on google and trying many different things).  Not sure what is the best way to report them or give you my installation notes, but I'll start with the very basic one.  

In your install instruction, it looks like there is a typo as my raspberry pi states it can't find /etc/system/systemd directory.  Shouldn't it be "/etc/systemd/system"?  I see this directory.

sudo cp gymnasticon.service /etc/system/systemd (my raspberry can't find it)
sudo cp gymnasticon.service /etc/systemd/system (my raspberry goes to that directory)

This is my first time using github or any programming stuff so please forgive me if I am doing something right.

ptx2, 2020-08-15:
Indeed that is a typo. Thanks for reporting this.

(And yes you are doing it right!)

Re: the best way to report other issues you ran into. If it's general notes or lots of small/related things, dropping it all in a single issue probably makes sense. But any way you want to report it will be appreciated!

ptx2, 2020-08-15:
Fixed in 7233a56d37956ed730d6de7260ad9210501f2474.

closed 2020-08-15 by ptx2