ptx2/gymnasticon #7: Guide for installation on fresh Raspberry Pi

ptx2, 2020-08-15:
It would be useful to have a step-by-step doc for getting up and running from a fresh Raspberry Pi

ptx2, 2020-08-15:
The steps (from memory) are:

* sd-card creation
* creating `wpa_supplicant.conf` and `ssh` files on sd-card for headless operation
* installing dependencies (nodejs 12.x, libudev-dev)
* installing gymnasticon
* upgrading gymnasticon

Check docs/guides for other Raspberry Pi projects like PiHole, RetroPie, etc. to see how they handle this.

rmaster78, 2020-08-15:
I can try to take a stab at this as I went through some issues while installing it.  To start off, the below is my install troubleshooting log that I created after failing to install it a few times.  I did this so I could remember what I was doing and read the error messages and log files more carefully.  I don't know if all the steps are required, but I know if anyone follows these steps they should get the same error and eventually solve it and get it installed.  I'll see how quickly I can make a cleaned-up version.

[install log.txt](https://github.com/ptx2/gymnasticon/files/5079198/install.log.txt)

rmaster78, 2020-08-17:
Install instruction using MacOS

	•	format sd card in disk utility
	•	download raspberry pi os lite from https://www.raspberrypi.org/downloads/raspberry-pi-os/
	•	install pi os lite https://www.raspberrypi.org/documentation/installation/installing-images/
	⁃	I used direct image copy as my MacOS is too old to use their installer
	•	enable usb connection https://www.tomshardware.com/reviews/raspberry-pi-headless-setup-how-to,6028.html 
	⁃	I only did this as I couldn’t get the headless wifi ssh connection to work for some reason even tho I tried multiple times
	⁃	create a blank ssh file (no extension) and copy it into sd card in the parent directly
	⁃	follow the USB connection instruction inside the link by editing 2 txt files in the SD card
	•	insert the SD card into pi zero w and connect usb cable to micro-usb port labeled USB not PWR as you need to use data usb connection to access ssh
	•	connect the other end of the usb cable to laptop and wait for pi zero w to boot up
	⁃	wait for flashing light to be stable, takes few minutes (est. 3 to 5min)
	•	open terminal and type ssh pi@raspberrypi.local
	•	⁃	if you get connection error
	⁃	make sure to use usb cable that is capable of transferring data, I couldn’t connect at first due to issues with usb cables that were supplying power only
	⁃	if you try to log in too early, before pi is fully boot up or forgot to copy the blank ssh file into SD card, you’ll get connection refused error
	•	it will ask you about authenticity of host and if you want to connect, say yes
	•	it will ask for password, type the default password raspberry
	•	setup wi-fi https://www.raspberrypi.org/documentation/configuration/wireless/wireless-cli.md
	⁃	I used raspi-config tool (type sudo raspi-config)
	⁃	goto localization options and scroll down until you see your country and press enter
	⁃	goto network options and wireless LAN to enter your wifi name and password
	⁃	press tab and high light finish and press enter
	⁃	your pi zero w should now be on your network, you can test this by using another laptop or shutdown the pi zero w (don’t just pull the cable as this could damage the raspberry pi, use command sudo poweroff and wait until flashing light turns off) and power it back up using PWR usb port connection on the pi zero w
	•	install node.js https://www.instructables.com/id/Install-Nodejs-and-Npm-on-Raspberry-Pi/
	⁃	gymnastics’s GitHub has link to the version (https://unofficial-builds.nodejs.org/download/release/v12.18.3/)
	⁃	extracting takes a while so be patient
	•	 run below command in terminal so you can give write access to that folder, it will give you error if you don’t during install 
	⁃	sudo chown -R $USER /usr/local/lib/node_modules 
	•	install missing library
	⁃	sudo apt-get install libudev-dev
	•	run below command to install gymansticon
	⁃	sudo npm install -g gymnasticon --unsafe-perm=true --allow-root
	•	once install is completed run below command to make sure it is working
	⁃	sudo setcap cap_net_raw+eip $(eval readlink -f $(which node))
	⁃	You should see the program trying to connect to the bike, spin the pedal a few times and you should see the program connect and start showing you watt and cadence info
	•	Create gymnasticon.service file to automatically run this program when pi boots up
	⁃	sudo nano /lib/systemd/system/gymnasticon.service
	⁃	copy paste text from https://github.com/ptx2/gymnasticon/blob/master/deploy/gymnasticon.service 
	⁃	sudo cp gymnasticon.service /etc/systemd/system
	⁃	sudo systemctl enable gymnasticon
	⁃	sudo systemctl start gymnasticon
	•	Reboot the device and turn on Zwift and it should see the power and cadence sensor

ptx2, 2020-08-25:
Thank you for writing all of this up. I'm sorry it took 19 steps!

I just wanted to update the ticket to point out there is now a custom SD card image available with Gymnasticon pre-installed and configured. For anyone with an SD card reader/writer on their computer, it should hopefully be a lot easier to get started.

https://github.com/ptx2/gymnasticon#quick-start-install-gymnasticon-sd-card-image

closed 2021-01-23 by ptx2