ptx2/gymnasticon #8: Intermittent power drop in Zwift

rmaster78, 2020-08-17:
I've been running Zwift for 2 days now and it happened both times pretty clearly where the power shown in Zwift companion app would sometimes show "---" which indicates zero watts or not being able to read watts.  The normal Zwift app that is on my Ipad uses 3 sec average number and I see the power number drop significantly even tho I was still pedaling at the same resistance and cadence.  Due to this, I would get pushed back into the group and would have to hustle back up to the group.

output from _journalctl -u gymnasticon -f_ command is below.  I road the bike from 10pm to 11pm last night 8/16/20, but the log file is at 11:10pm?

-- Logs begin at Sat 2020-08-15 21:07:48 PDT. --
Aug 16 23:10:38 raspberrypi gymnasticon[14829]:     at Hci.emit (events.js:315:20)
Aug 16 23:10:38 raspberrypi gymnasticon[14829]:     at Hci.onSocketData (/usr/local/lib/node_modules/gymnasticon/node_modules/@abandonware/noble/lib/hci-socket/hci.js:441:12)
Aug 16 23:10:38 raspberrypi gymnasticon[14829]:     at BluetoothHciSocket.emit (events.js:315:20)
Aug 16 23:10:38 raspberrypi systemd[1]: gymnasticon.service: Main process exited, code=exited, status=1/FAILURE
Aug 16 23:10:38 raspberrypi systemd[1]: gymnasticon.service: Failed with result 'exit-code'.
Aug 16 23:10:39 raspberrypi systemd[1]: gymnasticon.service: Service RestartSec=1s expired, scheduling restart.
Aug 16 23:10:39 raspberrypi systemd[1]: gymnasticon.service: Scheduled restart job, restart counter is at 7.
Aug 16 23:10:39 raspberrypi systemd[1]: Stopped Gymnasticon.
Aug 16 23:10:39 raspberrypi systemd[1]: Started Gymnasticon.
Aug 16 23:10:43 raspberrypi gymnasticon[15008]: [2020-08-17T06:10:43.594Z] connecting to bike...

rmaster78, 2020-08-18:
Not sure if this is related, but I reinstalled gymnasticon (in effort to create a clean "how to install" documentation).  I logged the terminal info when first starting up.

Few observations... 
1) if the cadence drops below 50 rpm with resistance set somewhere in the middle to high, the power reading is ZERO
2) got a "noble warning: unknown handle 65 disconnected message, but I am not sure what this is... 

[2020-08-18T15:37:01.573Z] connecting to bike...
[2020-08-18T15:37:21.466Z] bike connected ec:00:a2:d3:49:a3
[2020-08-18T15:37:21.652Z] received stats from bike [power=23W cadence=61rpm]
[2020-08-18T15:37:21.669Z] pedal stroke [timestamp=1597765041664 revolutions=1 power=23W]
[2020-08-18T15:37:22.649Z] received stats from bike [power=23W cadence=61rpm]
[2020-08-18T15:37:22.657Z] pedal stroke [timestamp=1597765042656 revolutions=2 power=23W]
[2020-08-18T15:37:23.639Z] received stats from bike [power=23W cadence=59rpm]
[2020-08-18T15:37:23.674Z] pedal stroke [timestamp=1597765043674 revolutions=3 power=23W]
[2020-08-18T15:37:24.644Z] received stats from bike [power=23W cadence=63rpm]
[2020-08-18T15:37:24.652Z] pedal stroke [timestamp=1597765044651 revolutions=4 power=23W]
[2020-08-18T15:37:25.606Z] pedal stroke [timestamp=1597765045605 revolutions=5 power=23W]
[2020-08-18T15:37:25.634Z] received stats from bike [power=23W cadence=63rpm]
[2020-08-18T15:37:26.559Z] pedal stroke [timestamp=1597765046558 revolutions=6 power=23W]
[2020-08-18T15:37:26.696Z] received stats from bike [power=23W cadence=65rpm]
[2020-08-18T15:37:27.483Z] pedal stroke [timestamp=1597765047482 revolutions=7 power=23W]
[2020-08-18T15:37:27.671Z] received stats from bike [power=23W cadence=65rpm]
[2020-08-18T15:37:28.407Z] pedal stroke [timestamp=1597765048406 revolutions=8 power=23W]
[2020-08-18T15:37:28.646Z] received stats from bike [power=23W cadence=65rpm]
[2020-08-18T15:37:29.331Z] pedal stroke [timestamp=1597765049330 revolutions=9 power=23W]
[2020-08-18T15:37:29.696Z] received stats from bike [power=23W cadence=65rpm]
[2020-08-18T15:37:30.255Z] pedal stroke [timestamp=1597765050254 revolutions=10 power=23W]
[2020-08-18T15:37:30.671Z] received stats from bike [power=23W cadence=65rpm]
[2020-08-18T15:37:31.179Z] pedal stroke [timestamp=1597765051178 revolutions=11 power=23W]
[2020-08-18T15:37:31.645Z] received stats from bike [power=23W cadence=64rpm]
[2020-08-18T15:37:32.116Z] pedal stroke [timestamp=1597765052116 revolutions=12 power=23W]
[2020-08-18T15:37:32.886Z] received stats from bike [power=23W cadence=64rpm]
[2020-08-18T15:37:33.055Z] pedal stroke [timestamp=1597765053054 revolutions=13 power=23W]
[2020-08-18T15:37:33.672Z] received stats from bike [power=23W cadence=65rpm]
[2020-08-18T15:37:33.978Z] pedal stroke [timestamp=1597765053977 revolutions=14 power=23W]
[2020-08-18T15:37:34.647Z] received stats from bike [power=23W cadence=65rpm]
[2020-08-18T15:37:34.902Z] pedal stroke [timestamp=1597765054901 revolutions=15 power=23W]
[2020-08-18T15:37:35.696Z] received stats from bike [power=24W cadence=65rpm]
[2020-08-18T15:37:35.825Z] pedal stroke [timestamp=1597765055824 revolutions=16 power=24W]
[2020-08-18T15:37:36.672Z] received stats from bike [power=25W cadence=65rpm]
[2020-08-18T15:37:36.748Z] pedal stroke [timestamp=1597765056747 revolutions=17 power=25W]
[2020-08-18T15:37:37.647Z] received stats from bike [power=26W cadence=65rpm]
[2020-08-18T15:37:37.671Z] pedal stroke [timestamp=1597765057671 revolutions=18 power=26W]
[2020-08-18T15:37:38.595Z] pedal stroke [timestamp=1597765058595 revolutions=19 power=26W]
[2020-08-18T15:37:38.697Z] received stats from bike [power=26W cadence=63rpm]
[2020-08-18T15:37:39.548Z] pedal stroke [timestamp=1597765059547 revolutions=20 power=26W]
[2020-08-18T15:37:39.673Z] received stats from bike [power=26W cadence=63rpm]
[2020-08-18T15:37:40.501Z] pedal stroke [timestamp=1597765060500 revolutions=21 power=26W]
[2020-08-18T15:37:40.647Z] received stats from bike [power=26W cadence=60rpm]
[2020-08-18T15:37:41.502Z] pedal stroke [timestamp=1597765061501 revolutions=22 power=26W]
[2020-08-18T15:37:41.697Z] received stats from bike [power=25W cadence=60rpm]
[2020-08-18T15:37:42.503Z] pedal stroke [timestamp=1597765062502 revolutions=23 power=25W]
[2020-08-18T15:37:42.673Z] received stats from bike [power=25W cadence=60rpm]
[2020-08-18T15:37:43.504Z] pedal stroke [timestamp=1597765063504 revolutions=24 power=25W]
[2020-08-18T15:37:43.647Z] received stats from bike [power=24W cadence=60rpm]
[2020-08-18T15:37:44.506Z] pedal stroke [timestamp=1597765064506 revolutions=25 power=24W]
[2020-08-18T15:37:44.697Z] received stats from bike [power=23W cadence=57rpm]
[2020-08-18T15:37:45.559Z] pedal stroke [timestamp=1597765065559 revolutions=26 power=23W]
[2020-08-18T15:37:45.673Z] received stats from bike [power=23W cadence=57rpm]
[2020-08-18T15:37:46.613Z] pedal stroke [timestamp=1597765066612 revolutions=27 power=23W]
[2020-08-18T15:37:46.647Z] received stats from bike [power=22W cadence=57rpm]
[2020-08-18T15:37:47.666Z] pedal stroke [timestamp=1597765067665 revolutions=28 power=22W]
[2020-08-18T15:37:47.698Z] received stats from bike [power=22W cadence=60rpm]
[2020-08-18T15:37:48.667Z] pedal stroke [timestamp=1597765068667 revolutions=29 power=22W]
[2020-08-18T15:37:48.682Z] received stats from bike [power=22W cadence=60rpm]
[2020-08-18T15:37:49.648Z] received stats from bike [power=22W cadence=60rpm]
[... cut ...]
[2020-08-18T15:38:06.244Z] pedal stroke [timestamp=1597765086243 revolutions=41 power=0W]
noble warning: unknown handle 65 disconnected!
[2020-08-18T15:38:06.646Z] received stats from bike [power=0W cadence=42rpm]
[2020-08-18T15:38:07.649Z] received stats from bike [power=0W cadence=42rpm]
[2020-08-18T15:38:07.672Z] pedal stroke [timestamp=1597765087671 revolutions=42 power=0W]
[2020-08-18T15:38:08.625Z] received stats from bike [power=0W cadence=37rpm]
[2020-08-18T15:38:09.294Z] pedal stroke [timestamp=1597765089293 revolutions=43 power=0W]

rmaster78, 2020-08-19:
After the ride last night, I pulled the terminal log information and indeed the bug where power drop to ZERO still exists.  It seems to happen every 2 to 3 min or so.  It seems like the code that was developed to mitigate this isn't working or some other issue exists?  

I am attaching raw terminal log file + excel file I used to create the chart.  Hope dev or someone can look into this and hopefully have a solution as this affects group rides quite a bit.

[gymnasticon zwift ride log 20200818.xlsx](https://github.com/ptx2/gymnasticon/files/5099403/gymnasticon.zwift.ride.log.20200818.xlsx)

[gymnasticon zwift ride log 20200818.txt](https://github.com/ptx2/gymnasticon/files/5099408/gymnasticon.zwift.ride.log.20200818.txt)

[image]

ptx2, 2020-08-20:
Thanks for the very thorough report and sorry you had to go through that. The zero power fix indeed is not enabled, or in fact, even in the code on the npm release :) I just added it and released version 1.0.3, give that a shot and let me know how you go.

I am strapped for time right now so haven't had a chance to look closely at the data you posted yet but will take a look later. I am curious to see how similar it is to what I've seen on my bike.

In version 1.0.3 the fix is always-on so all you should need to do is upgrade and restart the process with `systemctl`. If you want to see the impact of the fix, you can add an environment variable `NODE_DEBUG=flywheel` and it will log a message every time the fix activates.

To enable that environment variable running under systemd, you can add the following line to your `/etc/systemd/system/gymnasticon.service` file under the `[Service]` block:

```
Environment=NODE_DEBUG=flywheel
```

Then restart with:

```
sudo systemctl daemon-reload
sudo systemctl restart gymnasticon
```

Every time the fix activates you should see a message in the `journalctl` logs like this:

```
FLYWHEEL 7912: *** replaced zero power with previous power 92
```

rmaster78, 2020-08-20:
Hello P!

Thanks for quickly uploading the new version!  Just typed in the npm install command and it updated 1 packet and version shows 1.0.3 now.  I'll be going on another Zwift training ride today, so I'll report my findings after the ride here.  If the bug is resolved, I'll close the issue.

Once again thanks for getting this done during your tight schedule.  Really grateful for your time spent on creating this SW.  I am really enjoying the Zwift experience so far (even with the bug and I am sure it will be even better after the bug fix) and I am sure others feel the same way.

rmaster78, 2020-08-21:
Just finished the ride today and the issue is fixed!  It was so much easier to keep up with the group also!  Thanks for the release!

closed 2020-08-21 by rmaster78