ptx2/gymnasticon #9: New Feature Request: Decreasing power by specific % or W

rmaster78, 2020-08-24:
It isn't a bug or issue with the SW, but I think it would be a nice feature to have.

Myself and other's who own spin bikes are noticing that you can generate slightly more power at a given effort than a calibrated smart trainer.  Wondering if a feature can be added to Gymnasticon to reduce this power by a certain % or by certain Watt.  

To prevent abuse of this feature, perhaps make it so that only a negative value can be entered?

ptx2, 2020-08-25:
This would certainly be easy to add.

I'm wondering if this could also have something to do with the value of the `Sensor Location` characteristic that Gymnasticon reports to Zwift. Currently we say that the power sensor is taking measurements at the `Rear Hub` but perhaps it should be saying crank or pedal. I had to guess what to use here because I couldn't find any specifics on where LifeFitness measures power during their initial factory calibration.

I actually got a set of power meter pedals for my road bike recently and have been meaning to put them on the spin bike to see how the power measurement compares. I'll try this tomorrow and see what I get.

Do you have data or a rough idea of how off your readings are in % or W?

rmaster78, 2020-08-25:
Thanks for considering it!  

**TLDR:** 
1) Reddit guy has more accurate data set and he found 4.6% difference between his pedal power meter vs. flywheel bike
2) Mine is about 3 to 5W higher
3) My rough unscientific test comparing between TACX T2 trainer and flywheel bike was about 20W
4) Hoping your test reveals more/better data

**Entire comment:** 
The best data point I can get you is from one of the reddit guys in [this thread](https://www.reddit.com/r/FlywheelAnywhere/comments/iay5dy/gymnasticon/)... He did two FTP test where one he used his old spin bike with Garmin Vector pedals (back in Jan) and the other with Gymnasticon on Flywheel bike (8/22/2020) 

"FYI, I did the FTP test. 4.6% higher than the joroto with the Vectors. Honestly I'm pretty happy with that."

[new ftp on flywheel bike](https://drive.google.com/file/d/1v44QO4Q0jHChejIKJC5mrgshrF40-M36/view) 
[old ftp on old bike with powermeter pedal](https://drive.google.com/file/d/1poLRnrfgk13NNdtcyIxacBPnZ8IQhC1V/view) 

if you look at his lactate normalized power (at the bottom of screenshot) and multiply 95% (since he is doing 20 min FTP test), he gets 227 on new bike vs. 217 on old bike with pedal power meter.  About 10W of difference or 4.6% higher on flywheel bike with gymnasticon. 

If you continue reading, I did a quick comparison of his Flywheel bike vs mine where there is a difference of about 5W difference at lower cadence and 3W difference at higher cadence at 50% resistance.  

I also did a quick test ride where I borrowed my friend's TACX T2 smart trainer and set it next to the spin bike.  I jumped back and forth between the two bikes to see what the power output was at a given heart rate, resistance felt on my legs, and rpm.  It wasn't the most scientific as I just took screenshots (and the power on the TACX would drop drastically if I let ANY pressure off the pedal).  Anyways, this inaccurate comparison found that there is about 20W difference... 

I hope your results will show more data points.  BTW I recalibrated the bike using [derailleur program](https://github.com/benwithjamin/Derailleur) for our bikes.  It supposedly follows what flywheel bike's official software used to do to calibrate the bike.

ptx2, 2020-08-26:
I haven't had a chance to compare with the power meter pedals yet but I just released 1.0.4 with two new CLI options: `--power-scale` and `--power-offset`

Example usage:

```
# Decrease by 10%
gymnasticon --power-scale=0.9

# Decrease by 10 watts
gymnasticon --power-offset=-10

# Increase by 20%
gymnasticon --power-scale=1.2

# Increase by 20 watts
gymnasticon --power-offset=20
```

rmaster78, 2020-08-26:
Thanks for the quick update!  Just did a quick test on power scale mode and it seems to work!  Yah!  Not sure if running on scale or offset is better, but hopefully when you get your test done more will be revealed.  

As for running the command, it seems like I need to stop the current process and run it in terminal with the below command
`pi@raspberrypi:~ $ gymnasticon --power-scale=0.95`

I tried typing the below command so I can try to enable it all the time and don't have to have my computer open to run it, but it doesn't seem to work... it gives me an error "systemctl: unrecognized option '--power-scale=1'"
`pi@raspberrypi:~ $ sudo systemctl start gymnasticon --power-scale=0.95`

How can I enable it so it auto starts with this power scale or offset setting on?

ptx2, 2020-08-26:
Ah yeah so you just need to add it to the `/etc/systemd/system/gymnasticon.service` file on the `ExecStart=` line.

e.g. if you currently have:

```
ExecStart=/usr/local/bin/gymnasticon
```

Just change that to:

```
ExecStart=/usr/local/bin/gymnasticon --power-scale=0.95
```

Then run:

```
sudo systemctl daemon-reload
sudo systemctl restart gymnasticon
```

(The `daemon-reload` tells systemd that you modified a .service file and that it should take another look at it. And then `restart` actually restarts gymnasticon with the new options)

rmaster78, 2020-08-26:
Thanks! Got it running in auto mode again!

I am curious to see what you'll find after FTP test.  Let me know once you get enough time and motivation to get it done :).

rmaster78, 2020-09-04:
closing ticket as it is working!

closed 2020-09-04 by rmaster78