ptx2/gymnasticon #21: Troubleshooting for Peloton Cabling

jeremydk, 2020-10-14:
Creating an umbrella issue to discuss, rather than using PR #12

miked4499, 2020-10-14:
[image]
[image]
[image]

as requested

jeremydk, 2020-10-14:
I can't read the labeling on the contacts on the serial -- Can you check connectivity between the RX pad on the serial and the ring on the 3.5mm?  Additionally, confirm that TX isn't connected.  The behavior you're describing is a missed crossover/incorrectly wired cable, which is why I tried to push people toward the RCA (non-soldering) solution described in #12 because it's visually very easy to setup and confirm.  I don't think this is the splitter.

miked4499, 2020-10-14:
[image]

In this example, white appears to be RX and black is GND with TX being disconnected altogether. I'm sorry but i don't have anything to test the ring connectivity with.

I originally ordered the parts list from the original photo and found that i only got 1 of the RCA cables, got impatient and decided to solder as I had everything for that. I could also just get the 2nd RCA adapter and rewire the serial cable if need be.

jeremydk, 2020-10-14:
Without a multimeter, I'm not sure what I can advise from here. The RCA solution is recommended because it doesn't have any of the issues that soldering can bring to the table -- this could be anything from a short to the vendor deciding to change the colors of their wiring/where they're placed on the other end.

miked4499, 2020-10-14:
Understood. I'll go the RCA route. Thanks for the assistance!

notsureifkevin, 2020-10-15:
https://www.amazon.com/3-5mm-Stereo-Female-terminal-connector/dp/B077XPSKQD will also work.

miked4499, 2020-10-26:
Me again, switched to a RCA setup and am running into the same issue. Garmin device picks up and connects to the power meter, but no data is being sent. This is with a new un-soldered serial cable, and the RCA connectors shown in the attached photo. Have an actual ride running while testing. Any advice is appreciated. Is there anything different I need to do within the gymnasticon code? I'm literally, writing the recent image to a card, inserting it into the pi, powering on the pi, and launching a ride. 
[image]

jeremydk, 2020-10-26:
What command line are you running gymnasticon with? I believe the default
behavior is still set for Flywheel. You’ll need to pass the proper
arguments to the application for it to be in the right mode.

miked4499, 2020-10-26:
> What command line are you running gymnasticon with? I believe the default behavior is still set for Flywheel. You’ll need to pass the proper arguments to the application for it to be in the right mode.
> […](#)

I'm running it in its default mode. No changes are being made to the code. This is likely the issue. Is there documentation anywhere saying what to do to enable a peloton?

jeremydk, 2020-10-26:
The CLI options are documented here:

https://github.com/ptx2/gymnasticon

Please use the --bike peloton option and see if that resolves your issue.

Ashalls, 2021-01-13:
@miked4499 I'm facing a simular issue, I've checked the logs and the USB is picking up the peloton bike, im just not getting any readout. Did you manage to get it sorted in the end?

jeremydk, 2021-01-13:
Can you provide a log from Gymnasticon (including the lines as you start the application)?

What screen are you on on the Bike?

barrettec777, 2021-02-10:
I think I might be having a similar issue. Here's my gymnasticon log (from my raspberry pi zero WH).

```
Feb 09 20:31:21 gymnasticon systemd[1]: Started Gymnasticon.
Feb 09 20:31:33 gymnasticon gymnasticon[269]: [2021-02-10T01:31:33.240Z] connecting to bike...
Feb 09 20:31:33 gymnasticon gymnasticon[269]: [2021-02-10T01:31:33.317Z] bike connected /dev/ttyUSB0
Feb 09 20:31:33 gymnasticon gymnasticon[269]: [2021-02-10T01:31:33.355Z] no ANT+ stick found
```
Gymnasticon connects just fine to Zwift on my phone/ipad. But I see nothing more in the log when I pedal, and no response in Zwift. When I disconnect from Zwift, I see this (which may very well be expected behavior):

`Feb 10 09:22:14 gymnasticon gymnasticon[269]: noble warning: unknown handle 64 disconnected!`

I have the same RCA cabling setup as shown in the first issue as well as in this one, but I use a UGREEN USB to MicroUSB adapter to connect the USB-Serial decoder to the pi zero. I have a regular Pi 4 coming in the mail in a couple days, so I'll give that a try as well. 

I do have a multimeter if there's something you'd like me to check.

jeremydk, 2021-02-11:
I'd be happy to help take a look.  For the sake of not spamming folks previously on this topic, I'm going to go ahead and close this issue and open a new issue to help dig.

closed 2021-02-11 by jeremydk