ptx2/gymnasticon #22: Using Zwift sideloaded on to Peloton bike?

vodlogic, 2020-10-15:
I understand that for the Peloton bike to broadcast the power/cadence, a Peloton class must be loaded ready to start or in progress. Do you know if the Peloton app runs happily in the background, continuing to broadcast data so that we can use Zwift sideloaded on to the Peloton tablet screen or would we need to use another tablet for zwift? If not happy to give it a try when my cables arrive.

notsureifkevin, 2020-10-15:
It's possible to command the sensor directly and bypass using the Peloton service. You can _probably_ start a "Just Ride" session and swap it to the background, but I found the Peloton tablet entirely too underpowered to run Zwift so I just run it on a different device now.

https://github.com/notsureifkevin/peloton-power/blob/master/index.js#L36-L52

You can drop this code into the peloton bike service, and reverse the TIP/RING on the serial cable (both need to be connected), and you shouldn't even need the Peloton service running on the tablet.

vodlogic, 2020-10-15:
Thanks I'll give both approaches a go and if the latter works I'll consider raising a PR to call the setInterval as a cli option

vodlogic, 2020-10-22:
#24

PHOLAN, 2021-02-03:
Would it be possible to fully disconnect the Peloton Head Unit? Have the Raspberry send the wake up message to the bike upon a bluetooth device connection to the Raspberry, such from Zwift?

kevtainer, 2021-02-03:
@PHOLAN yes, it would be possible, #24 can already command the sensor package without the H/U, but you'd need to write a hook that would determine if a BT connection was established to start the polling. I sold my Peloton months ago, so I don't have any interest in the project now, but good luck if you decide to try and make that work.

ptx2, 2021-02-14:
@PHOLAN This works now in v1.3.0. The wiring shown here: https://github.com/ptx2/gymnasticon/pull/12#issuecomment-696345309 needs two modifications: 1) disconnect the tablet and 2) connect the disconnected RCA cable. Good luck!

closed 2021-02-14 by ptx2