ptx2/gymnasticon #23: Similar project qdomyos-zwift

cagnulein, 2020-10-21:
I found you searching on github, i started a similar project some weeks ago 😄 

https://github.com/cagnulein/qdomyos-zwift
[image]

ptx2, 2020-11-17:
Nice one!

closed 2020-11-17 by ptx2

igeo, 2025-07-09:
I have a FlyWheel studio bike instead of FlyWheel Home Bike. I am wondering if anybody has successfully connect to the studio bike?
Thank you for the great work.

[image]

[image]

[image]

igeo, 2025-07-17:
@cagnulein thank you for the response. At this time, I don't know if this bike broadcast BLE or not. I tried nrfconnect  and scan doesn't show anything like a bike.

cagnulein, 2025-07-17:
Show me a scan and I will double check it