ptx2/gymnasticon #25: Garmin devices unable to pick up data from Peloton

miked4499, 2020-10-28:
Wondering if anyone has any advice regarding connecting to Garmin devices? I have both a Fenix 6 and an Edge 830 which both show no readings for a power meter, yet they can find and connect to the Gymnasticon device. I’m able to connect my MacBook to the device and run Zwift, which leads me to believe there might be something on the Garmin side of things.

Thanks for any advice given

GMac22, 2020-12-31:
I have the same issue after trying this out last week. Using Raspberry Pi4 with the image. I can connect to the BT MAC on my Forerunner 935 as a power meter, but never get any power readings on the watch. 

When SSH into pi, and issue the command below, I can see power updating/matching the Peloton bike, so the Pi is picking up the wattage.
journalctl -u gymnasticon -f

Maybe it is Garmin side issue with a BT power meter. I saw some older posts on Garmin Dev forms about BT power meter not supported (although was a couple years ago).  It looks like the test here might have been ANT+ per https://github.com/ptx2/gymnasticon/pull/12#issuecomment-704212073

So maybe for Garmin connection, it is about either Getting/Confirming Garmin and the BT power meter, or going the route of ANT+ on Pi instead?

cagnulein, 2020-12-31:
I guess it's similar to this one https://github.com/cagnulein/qdomyos-zwift/issues/91

GMac22, 2020-12-31:
Looks like it may be. 

For reference too, Fenix 5 and 935 I believe are same firmware (same with Fenix 6 and FR 945). Which both product pages seem show show support for cycling power over BT:

SPEED AND CADENCE SENSOR SUPPORT | yes (ANT+® and Bluetooth® Smart sensors)

Both product pages:
https://buy.garmin.com/en-US/US/p/641449#specs and https://buy.garmin.com/en-US/US/p/564291#specs

cagnulein, 2021-01-01:
@GMac22 as i said here https://github.com/cagnulein/qdomyos-zwift/issues/91#issuecomment-753311798 today i did a test with running speed and cadence sensor on my polar m430 and everything works well! So i guess it's an issue of fenix firmware. Maybe you should open a ticket on Garmin support?

ihaque, 2021-01-07:
At least my Garmin watch (the Venu) does not support the Bluetooth Cycling Power Profile; only the Cycling Speed and Cadence Profile, which only reports speed and cadence, not power.

Last I looked at gymnasticon, it uses CPP to report both power and speed, but that doesn't work with the Garmin I have. More details here: https://ihaque.org/posts/2021/01/04/pelomon-part-iv-software/#gotcha-1-cycling-speed-and-cadence-vs-cycling-power

I wrote a tool based on different hardware at https://github.com/ihaque/pelomon to do this. A similar approach could likely be backported here if it doesn't already exist.

ptx2, 2021-01-31:
I've just pushed https://github.com/ptx2/gymnasticon/pull/36 that might be of interest for getting Gymnasticon to send power and cadence data to Garmin devices.

GMac22, 2021-02-02:
#36 worked for me using ANT+

Tested with Peloton > Pi4 (with ASIN: B01M3VQP6Z from amazon for ant+ stick) > Garmin FR 935 watch

ptx2, 2021-02-03:
Released in 1.2.0.

closed 2021-02-03 by ptx2