ptx2/gymnasticon #26: Future Project? Auto adjust resistance based on Zwift gradient %

rmaster78, 2020-10-29:
I recall reading from the ptx article that a future project is to add an auto resistance feature in. While I found out that Flywheel Home Bike is pretty much a rebranded IC5 bike from Life Fitness.  I was able to contact them and get a brake calibration manual and how to remove the shroud of the bike to get access to all the internals.

Perhaps we can discuss what needs to be done to get this going or even just some ideas that people have.  Based on what I can tell, it looks like the magnet is attached to a spring that pulls it back and a cable that is tied to the resistance knob and pulls the magnet in and out towards the flywheel to control the resistance that we feel.  Seems to me like a derailleur type setup where there is sprint tension and cable pull to modulate it. There also seems to be some micro adjustment screw to move the magnet in or our in very small increments It also seems like the travel of the magnet isn't long so perhaps we can bypass the entire cable system and attach a small linear actuator to push the magnet in or out?

manual pdf is in my dropbox link below... 
https://www.dropbox.com/sh/17r79rrc0xxgg4x/AADAWeP16HcKbjVYCM8XRiVha?dl=0

dagronslayer, 2020-11-06:
If I find the time this weekend, I'll take my trainer apart and measure the cable pull weight and distance on the Bowden wire at diagram 1.6. It seems like that would be the first step in finding a suitable linear actuator.

Lukanoski, 2020-11-25:
I'm making a direct drive trainer from an old magnetic resistance unit I found in a junkyard. I plan to add some sort of stepper to control the resistance from the raspberry pi and it would be great it can be controlled from Zwift or other training software.
So far i have made a Node-Red flow to send data via udp to the bot from two reed switches on the RPI and a dashboard hosted on the RPI. This would be a great feature for cheap DIY smart trainer. Probably we can use the same data for DIY kickr "climb".

doudar, 2020-12-15:
https://github.com/doudar/SmartSpin2k

dagronslayer, 2021-02-27:
Bowden cable at diagram 1.6 distance from lowest to highest resistance is 24mm. The pull weight is something over 2 lbs. I'm guessing 3 lbs would be enough, but after looking at it. I think the easiest method would be to disconnect the bowden cable and replace the tension spring with a linear actuator. It would need to be about 60mm closed and extend to about 100mm. Something like this would probably work:
https://www.amazon.com/EElabper-Actuator-Electric-Waterproof-Mounting/dp/B08QFZ723Q
I would take better measurements before ordering anything.