ptx2/gymnasticon #28: Connection on PC

Lukanoski, 2020-11-25:
'm making a direct drive trainer from an old magnetic resistance unit I found in a junkyard. So far i have made a Node-Red flow to send data via udp to the bot from two reed switches on the RPI and a dashboard hosted on the RPI.
This setup works fine on android, but not working on PC. I can see the gymnasticon sensor as power sensor, but not as cadence. Also I can't connect to the RPI as a BLE peripheral, but it works in the Zwift app on android.

ptx2, 2020-11-25:
Interesting project! Let us know if you do a write-up ;-)

What OS/version/app are you running on the PC?

Some initial thoughts, assuming Windows/Zwift:

* Try open "Event Viewer" and look at the "Event Log" to see if there are any Bluetooth-related messages. Maybe there is a better tool for this I haven't used Windows in a long time.
* Check Zwift logs https://support.zwift.com/en_us/locating-log-files-BykKO63fr
* Try apps other than Zwift (e.g. Rouvy)
* Try nRF Connect app to see if you can connect/subscribe. It may provide more details on exactly what is failing https://www.nordicsemi.com/Software-and-tools/Development-Tools/nRF-Connect-for-desktop

Lukanoski, 2020-11-27:
It works on android with nRF-Connect app. I can't connect on any device with nRF-Connect-for-desktop. I guess that something is wrong with my Windows 10 PC setup or Bluetooth dongle. However it works good enough for me at the moment. 
I've tested with RTG Cycling and OneLap and it connects and works.

I planned to document the project but it's still early, and I haven't completed the hardware yet. will post a link when I have something.

closed 2020-11-27 by Lukanoski