ptx2/gymnasticon #29: Connected to Zwift but "0" for power

jerrygass, 2021-01-04:
Hello, 

I have a flywheel bike, and gymnasticon running on a pi zero w. 

Cadence is working properly on zwift, and power shows connected to gymnasticon, but wattage remains at 0 when pedaling. 

Thoughts on a remedy?

thank you.

ptx2, 2021-01-06:
Hi,

How high is the resistance set and how high is your cadence? The bike will report 0 watts until you reach a certain threshold, so if just pedaling slowly on low resistance a reading of 0 watts is normal.

If that's not it, it's possible the bike needs calibrating.

To calibrate the bike using a Mac computer you could try using the [Derailleur app](https://github.com/benwithjamin/Derailleur/issues/1#issuecomment-634384336).

On iOS/Android the ICG Training app may let you calibrate the bike, but I'm not sure.

I also have a calibration tool you could try, let me know if none of the above options work and I'll try to find some time to make it user-friendly and put it online.

Failing all of that, the next step would be to grab some Gymnasticon logs while you're riding at a decent effort so we can see what data the bike is sending and start to narrow down the cause.

To grab the logs:

1. [https://raspberrypi.stackexchange.com/a/57023](Enable WiFi and SSH) on your Raspberry Pi so you can login over the network (or plug in a HDMI monitor and USB keyboard and login from the console)
2. Login to the Pi  e.g. `ssh pi@gymnasticon.local` (password `raspberry`)
3. Monitor the gymnasticon logs `journalctl -u gymnasticon -f` while riding the bike

Live logs similar to this should appear while you ride:

```
[2020-08-18T15:37:01.573Z] connecting to bike...
[2020-08-18T15:37:21.466Z] bike connected ec:00:a2:d3:49:a3
[2020-08-18T15:37:21.652Z] received stats from bike [power=23W cadence=61rpm]
[2020-08-18T15:37:21.669Z] pedal stroke [timestamp=1597765041664 revolutions=1 power=23W]
[2020-08-18T15:37:22.649Z] received stats from bike [power=23W cadence=61rpm]
[2020-08-18T15:37:22.657Z] pedal stroke [timestamp=1597765042656 revolutions=2 power=23W]
[2020-08-18T15:37:23.639Z] received stats from bike [power=23W cadence=59rpm]
[2020-08-18T15:37:23.674Z] pedal stroke [timestamp=1597765043674 revolutions=3 power=23W]
[2020-08-18T15:37:24.644Z] received stats from bike [power=23W cadence=63rpm]
[2020-08-18T15:37:24.652Z] pedal stroke [timestamp=1597765044651 revolutions=4 power=23W]
```

If you paste some of those logs here, it will help pinpoint the cause.

Let us know if you get stuck or if any of the above happens to work for you, good luck!

jerrygass, 2021-01-06:
Thank you for the response.

I have tried pedaling at >70rpm cadence, with the resistance knob set at the halfway point, still no power appears on zwift.

I do not have a Mac, but can borrow one if need be to try Derailleur. I was not sure if the calibration through derailleur was specifically for use with the derailleur's resistance display, or more of a calibration of the bike electronics itself as would be done on a regular IC5 bike that had the factory console. 

Would you have more details on your alternative calibration tool? I am in no rush.

reiser4, 2021-01-11:
hello, I see the same. I can see the wattage with ipbike on android phone, but zwift reports "no data". I am using bot bike with valid values:

```
[2021-01-11T19:24:44.495Z] pedal stroke [timestamp=1610393084493 revolutions=226 power=56W]
[2021-01-11T19:24:44.917Z] received stats from bike [power=56W cadence=60rpm]
[2021-01-11T19:24:45.494Z] pedal stroke [timestamp=1610393085493 revolutions=227 power=56W]
[2021-01-11T19:24:45.923Z] received stats from bike [power=56W cadence=60rpm]
[2021-01-11T19:24:46.494Z] pedal stroke [timestamp=1610393086493 revolutions=228 power=56W]
[2021-01-11T19:24:46.926Z] received stats from bike [power=56W cadence=60rpm]
[2021-01-11T19:24:47.494Z] pedal stroke [timestamp=1610393087493 revolutions=229 power=56W]
```

can I help to diagnose this? thanks.

reiser4, 2021-01-12:
Also RGT cycling can receive the power value. Maybe the issue is with having both power and cadence in the same device?

reiser4, 2021-01-12:
Zwift log says:
`[11:24:49] BLE  : Set notify failed`

hcoupe67, 2021-01-14:
Hello I'm having the same issue - just got the bike and was hoping to get it working with Zwift.

hcoupe67, 2021-01-14:
I just tried the ICG app on connect to it via BLE scanner and I get RPM and Calories but no Watts

DesEagle1220, 2021-01-18:
This is happening to me too, recently got Gymnasticon running, it gets cadence right and picks up every revolution but power is always 0 W.

reiser4, 2021-01-18:
I have been able to fix https://github.com/paixaop/zwack for my purpose (obtain watts in zwift) but the same fix does not apply to Gymnasticon. Zwack is enough for my project so I did not spend time to fix also Gymnasticon, but if it can be useful I can share the patch that works on Zwack.

lpecan, 2021-01-18:
Just curious if everyone here reported 0W output all have a bike that has never been calibrated before (which seems to be *a* cause for that)

hcoupe67, 2021-01-18:
Got my bike working - 

Like Ptx2 stated you have to get your bike calibrated for it to work.  
That is the main issue - 
You need someone with a Mac and a later one at that,  when I used Derailleur app with a MacBook Pro 2011 it won't see the bike because at that time the Bluetooth from the Mac didn't support Bluetooth LE or BLE - when I used a Mac Air 2016 - everything worked great - the program stated the bike wasn't calibrated and once done everything was great !! 

Thank you for the people that took the time and wrote the code to unbrick the bike!!

Here is the Derailleur app - Link again !! 

https://github.com/benwithjamin/Derailleur

Take Care

cagnulein, 2021-01-18:
> Just curious if everyone here reported 0W output all have a bike that has never been calibrated before (which seems to be _a_ cause for that)

i confirm that without calibration you will have 0W ;) same on https://github.com/cagnulein/qdomyos-zwift

lpecan, 2021-01-19:
@hcoupe67 can you meet us on reddit to help some new users? Lots of people are buying the bike in the last couple weeks and having trouble doing the initial calibration. I don't have a mac and have never used derailleur (though I've recommended it to people in the situation). At least one person reports not having success. Can you meet people here and try to lend some experience?  https://www.reddit.com/r/FlywheelAnywhere/comments/kzq3ox/power_reading_0_watts/

hcoupe67, 2021-01-20:
Hello,
I followed the link - do you want me to respond on the very top of the link?

ptx2, 2021-01-20:
@lpecan @hcoupe67 @jerrygass I've just published a small WebBluetooth-based tool called FlyTest that allows you to calibrate and test the Flywheel bike using just a web browser (Chrome on Mac/Linux/Windows/Android/ChromeOS and WebBLE on iOS).

I've only been able test it so far on WebBLE for iOS and Chrome for Mac.

This is a good option for those who do not have access to a Mac to use the Derailleur app mentioned above.

You can try it out here: https://ptx2.net/apps/flytest/

Here's what it looks like:

[image]

hcoupe67, 2021-01-20:
Ptx2 – Your amazing !!
Thank you for all your hard work !!

Awesome

Hcoupe67

[flytest]<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F68594395%2F105112861-d1b52700-5a91-11eb-8822-73454b2a1a74.gif&data=04%7C01%7C%7Ce6eb5077451f4a1cd67108d8bce044a8%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637467018688869934%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=PdMzc7rpMu%2FRy5ngiNVZ5ZS37ls6PdvSn%2B4hgyGFidA%3D&reserved=0>

lpecan, 2021-01-20:
Hcoupe67

Thanks. When you get a chance, take a look at the video mfunez linked in
the reddit threadand see if he (or she) did anything different than you. In
the meantime I asked him or her to try ptx2's new tool. My bike has been
perfect for six months now thanks to gymnasticon. I'm just trying to pay it
forward by helping new people. A lot of people seem to have purchased the
bikes from grocery outlet in the last couple weeks. Hence the increased
interest.

hcoupe67, 2021-01-20:
Hello,

That looks right - maybe make sure the bike is not connected to Gymnasticon by turning it off and letting Bluetooth reset and then connect using the Mac - but I really like Ptx2 new program - just use your smart phone to connect - much quicker and easier!!

Good Luck out there and enjoy the bike.

mikalasf, 2021-01-20:
Pxt2 - you saved the day. This did the trick. Thanks! 

> @lpecan @hcoupe67 @jerrygass I've just published a small WebBluetooth-based tool called FlyTest that allows you to calibrate and test the Flywheel bike using just a web browser (Chrome on [...]

jerrygass, 2021-01-21:
@ptx2 - Got flytest working on my android phone. It was straighforward, and effective. Now power shoes in ZWIFT. Thanks so much!

ptx2, 2021-01-23:
Happy to hear it!

closed 2021-01-23 by ptx2