ptx2/gymnasticon #30: Ignore huge power spike reported by bike

dagronslayer, 2021-01-09:
I stopped pedaling for just a second today and instantly received all the power achievements--up to 1.21 gigawatts (You hit 1200 watts).

I'm not sure how you'd fix this other than ignoring ridiculously high values reported from the bike. Here is the entry from my daemon.log including a few rows before and after:

Jan  9 13:49:43 gymnasticon gymnasticon[644]: [2021-01-09T18:49:43.451Z] received stats from bike [power=166W cadence=79rpm]
Jan  9 13:49:43 gymnasticon gymnasticon[644]: [2021-01-09T18:49:43.888Z] pedal stroke [timestamp=1610218183886.0063 revolutions=505 power=166W]
**Jan  9 13:49:44 gymnasticon gymnasticon[644]: [2021-01-09T18:49:44.410Z] received stats from bike [power=16499W cadence=79rpm]
Jan  9 13:49:44 gymnasticon gymnasticon[644]: [2021-01-09T18:49:44.646Z] pedal stroke [timestamp=1610218184645.5 revolutions=506 power=16499W]**
Jan  9 13:49:45 gymnasticon gymnasticon[644]: [2021-01-09T18:49:45.371Z] received stats from bike [power=0W cadence=0rpm]
Jan  9 13:49:46 gymnasticon gymnasticon[644]: [2021-01-09T18:49:46.391Z] received stats from bike [power=0W cadence=0rpm]
Jan  9 13:49:47 gymnasticon gymnasticon[644]: [2021-01-09T18:49:47.351Z] received stats from bike [power=0W cadence=0rpm]

I'm surprised Zwift doesn't throw this out, it looks like they just cap the figure at 2000W

   <trkpt lat="-11.6489000" lon="166.9530760">
    <ele>10.4</ele>
    <time>2021-01-09T18:49:46Z</time>
    <extensions>
     <power>2000</power>
     <gpxtpx:TrackPointExtension>
      <gpxtpx:hr>159</gpxtpx:hr>
      <gpxtpx:cad>78</gpxtpx:cad>
     </gpxtpx:TrackPointExtension>
    </extensions>
   </trkpt>

ptx2, 2021-01-23:
Sorry for the slow reply and thanks for reporting this! That's one impressive pedal stroke :-)

I haven't seen this myself. It looks like this is the Flywheel bike?

We could certainly add something to filter this out. Do you have any more logs (the more the better) showing different occurrences of this issue? Or can you share a bit more about it, e.g. is it always for exactly one pedal stroke after you stop pedaling? Does it happen every time you stop pedaling? Is it always this exact value 16499W and if not what's the lowest false value you've seen so far?

doudar, 2021-01-24:
If it's any consolation, I've seen this exact same behavior from my Elite Direto.

dagronslayer, 2021-01-25:
> Sorry for the slow reply and thanks for reporting this! That's one impressive pedal stroke :-)
>
> I haven't seen this myself. It looks like this is the Flywheel bike?
>
> We could certainly add something [...]

No worries on the slow reply. It is a Flywheel bike. This is the only occurrence I've seen in 4-5 months of using Gymnasticon. I'll  try and replicate it again with some sudden deceleration. It's going to be a few days (I just spent a weekend on my mountain bike for the first time in ~4 months, I'm really saddle sore.)

dagronslayer, 2021-02-06:
I've tried on a couple of rides to reproduce this and haven't been able to yet. Maybe a one-time fluke, or pretty rare occurrence?

ptx2, 2021-02-12:
Thanks for trying to reproduce it.

Maybe a cosmic ray flipped a bit? :-)

16499W is only one bit different from 115W (flip bit 14 from 1 to 0). The sequence 166W=>115W=>0W actually seems plausible.

I'm tempted to leave this one until it happens again. What do you think?

dagronslayer, 2021-02-12:
I usually blame solar flares, which I guess is the same thing? haha

I agree, as the only known instance, it's either a fluke or very rare. not worth fixing unless someone is eventually able to confirm it.

closed 2021-02-14 by ptx2