ptx2/gymnasticon #31: Zwift on Windows without Companion app not working

volfyd, 2021-01-19:
Hi,

Has anyone gotten this to work successfully with Zwift on Windows without the Zwift companion app?

The companion app works for me (Peloton bike here), and both cadence and power are recognized. But occasionally I get disconnects and I wanted to try to connect directly to my PC.

When I try to connect this way, I can connect to "Gymnasticon 145" under Power but nothing under Cadence and I never get anything besides 0 watts and usually I get "No Signal" after about 5 seconds after it connects and displays zero watts.

I didn't find anything useful in the Zwift logs. Just this:
```
[18:09:29] Device List:

[18:09:29] ==========================================================

[18:09:29]                 Power: Gymnasticon 145 (0x12618591) [BLE]

[18:09:29] ==========================================================

```
I don't know what 145 is.

I tried updating the drivers for my intel bluetooth. I also tried 3 different bluetooth dongles (by disabling the built-in bluetooth before connecting the dongle).

ptx2, 2021-01-23:
Hi!

I haven't actually tried with Windows myself, but in theory it should work.

The first thing that comes to mind is Bluetooth signal. A simple test is to position the Raspberry Pi and Windows PC so that their Bluetooth antennas have direct line-of-sight and are only a few feet apart. Turn off any other devices in the area that may be causing interference.

Another thing that comes to mind is the device you use for the companion app may be stealing the connection. Even if you have the companion app closed it's possible the phone or tablet is staying connected. A simple test would be to shut that device off entirely before trying to connect with the Windows computer.

See this page for some examples of interfering devices (under "Wireless signal interference") and also some general BLE troubleshooting tips: https://support.zwift.com/en_us/ble-vs-ant-interference-troubleshooting-tips-ryUhOTnzH

I'll see if I can get my hands on a Windows PC to test. Do you have details about the Bluetooth adapters you tried (e.g. make/model) and also Windows version?

doudar, 2021-01-23:
I agree with @ptx2 . Zwift Companion (especially iOS) is **not** a team player. It steals every BLE FTMS device while it's open. Zwift doesn't need to be on the pairing screen either. Windows solo (turn off bluetooth on your Zwift Companion device) doesn't have those issues. 

see:
https://github.com/doudar/SmartSpin2k/issues/49

doudar, 2021-01-23:
I should add however that if that's 100% the case, Gymnasticon 145 shouldn't be showing up on your Zwift log. 

Make sure you are not pairing the Gymnasticon 145 in the Windows OS menu also as that causes problems. Only pair it from the Zwift app. If it is paired via the bluetooth icon on your windows desktop, forget/disconnect the device. That's really only for HID bluetooth Mice, keyboards, speakers, etc....

volfyd, 2021-01-29:
So I have tried three different bluetooth chips, the intel one in my computer with an external antenna, direct line of sight, and these two dongles, direct line of sight. All of these were with my cell phone's bluetooth **off**.

1. Computer: https://www.amazon.com/ASUS-ROG-B550-F-Motherboard-Addressable/dp/B088VSTS9H
2. dongle: https://www.amazon.com/Importer520-Bluetooth-Adapter-Raspberry-Compatible/dp/B00CG94OW8/ref=sr_1_1?dchild=1&keywords=csr8510&qid=1611948962&sr=8-1
3. dongle: https://www.amazon.com/gp/product/B009ZIILLI/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

Each time, I can connect to Gymnasticon 145 as Power, but not Cadence, and it shows 0 watts for a second before switching to No Signal.

I did not try to pair gymnasticon with the windows bluetooth menu and it does not show there either.

[image]
[image]

reiser4, 2021-01-29:
Hello, I had the same issue. So I tried "zwack" that did the same. I was able to fix zwack by changing the descriptor a bit and ended up using it because it's enough for my usage (I obtain the watts from an Arduino device).  
If it can help, I can share the changed that I needed to make to zwack in order to fix the "No signal" problem.

cagnulein, 2021-01-29:
@reiser4 yes! I think it could be helpful!

reiser4, 2021-01-30:
here is the code that works for me:

```
constructor() {
    super({
      uuid: '2A63',
      value: null,
/////// CHANGE 1: ADD 'read', 'write', 'writeWithoutResponse'
      properties: ['notify', 'read','write', 'writeWithoutResponse'],
      descriptors: [
        new Bleno.Descriptor({
					uuid: '2901',
					value: 'Cycling Power Measurement'
				}),
/////// CHANGE 2: DISABLE THIS DESCRIPTOR
//        new Bleno.Descriptor({
 //         // Client Characteristic Configuration
 //         uuid: '2902',
 //         value: Buffer.alloc(2)
 //       }),
        new Bleno.Descriptor({
          // Server Characteristic Configuration
          uuid: '2903',
          value: Buffer.alloc(2)
        })
      ]
    });
    this._updateValueCallback = null;  
  }
```

volfyd, 2021-01-30:
Huh, cool. So with the suggestion by @reiser4 I got Zwift to connect to Gymnasticon without the Companion app. The only thing I did was comment out the 2902 descriptor in cycling-power-measurement.js:

```
      properties: ['notify'],
      descriptors: [
//        new _bleno.Descriptor({
//          uuid: '2902',
//          value: Buffer.alloc(2)
//        }),
        new _bleno.Descriptor({
          uuid: '2903',
          value: Buffer.alloc(2)
        })
      ]
```

I haven't actually done a ride like this, just verified that the pairing screen shows reasonable values when I turn the Peloton crank half a turn.

This does seem a bit odd to me, because I thought 2902 is a required descriptor.

volfyd, 2021-01-30:
Also FulGaz is happier. Before it had trouble connecting and always showed 0 for cadence and power. Now it shows the correct power but the cadence gets stuck.

volfyd, 2021-01-30:
And with 2902 commented out, I can still connect via the Companion app.

volfyd, 2021-01-31:
I am not familiar enough with Bleno to be sure if this code is being called but at 

https://github.com/abandonware/bleno/blob/master/lib/hci-socket/gatt.js#L205

```
      if (properties & 0x30) { // notify or indicate
        // add client characteristic configuration descriptor

        handle++;
        var clientCharacteristicConfigurationDescriptorHandle = handle;
        this._handles[clientCharacteristicConfigurationDescriptorHandle] = {
          type: 'descriptor',
          handle: clientCharacteristicConfigurationDescriptorHandle,
          uuid: '2902',
          attribute: characteristic,
          properties: (0x02 | 0x04 | 0x08), // read/write
          secure: (secure & 0x10) ? (0x02 | 0x04 | 0x08) : 0,
          value: new Buffer([0x00, 0x00])
        };
      }

```

Perhaps the descriptor 2902 is getting created twice, and depending on some order of operations, 2902 gets updated to enable notifications with the companion app and when read back that works correctly, but on other apps the duplicate 2902 gets read, which is not enabled for notifications and so the devices get no data.

That fits with what I'm seeing where gymnasticon will be found but immediately no signal is returned.

ptx2, 2021-01-31:
@reiser4 @volfyd Nice find!

Funnily enough, I had to add that descriptor to get it working at one point. It was probably an older version of bleno and perhaps a old version of macOS (this was prior to the initial commit).

I just tested this change on all major platforms (recent macOS, Raspbian, and you have tested Windows) and it works just fine so I'll just go ahead and remove it.

chriselsen, 2021-01-31:
For me this fix actually isn't working. I'm still getting the same error from an Apple TV 4K & Samsung Tab S6 as client with Gymasticon running on either Raspberry (with external Bluetooth 4 stick) or on a Mac Mini with the build-in Bluetooth module. 

In all cases:
- Gymasticon only shows up as a Power source, but not Cadence source
- It goes in and out between Connected with 0 watts and No Signal

ptx2, 2021-01-31:
@chriselsen Bummer, ok. I don't have an Apple TV 4K or Samsung Tab S6 (or other Android device) handy to test.

* Is there anything in the Zwift log?
* What happens if you also remove the 2903 descriptor (in fact, just remove the `descriptors:` key entirely). And then `npm run build` and run gymnasticon again?

If no luck with the above, I think the next step is to start digging through logs. You can post them here or email them to me if you want and I can try to help. It would be good to see Bluetooth debug logs for both sides of the connection. On the Raspberry Pi start `btmon -w btsnoop.log` before starting Gymnasticon. On Samsung I think you can also generate a `btsnoop_hci.log` by enabling Bluetooth logging in the Developer menu. You can also turn on application-level debug logs by running Gymnasticon with `NODE_DEBUG=* DEBUG=*` in the environment.

Also can you share more details about the setups? e.g.

Mac Mini
System Information > Software > System Version
System Information > Hardware > Model identifier
System Information > Hardware > Bluetooth > Hardware, Features, and Settings > everything except name/mac address

Raspberry Pi
External Bluetooth stick make/model

chriselsen, 2021-01-31:
Looks like my external Bluetooth stick is Bluetooth 4.0 only, which could be the cause for this. 
This is how it shows up: 
ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)

Ordered a Raspberry 4 with build-in Bluetooth to rule out that this is the issue. Will report back once it arrives to see if that was the culprit.

ptx2, 2021-02-01:
I just realized the Bluetooth requirements in the README were from a time when only the Flywheel bike was supported. Just updated it to say a Bluetooth 4.1+ multi-role adapter is required but only if you're using a Bluetooth LE bike (e.g. Flywheel). For testing with the bot or using a Peloton with the serial cable a 4.0 adapter should work in theory. So it could be the adapter but maybe there's still a software fix needed here. The Pi 4 test should indeed help narrow it down!

chriselsen, 2021-02-01:
Tried it again: Now it's working with the Apple TV 4K (which I actually use for Zwift). It picks up power and cadence correctly from the Bot. 
Only the two Samsung Tablets are still having issues. 

We can close this ticket. 

I'll nevertheless test with the Raspberry Pi 4 if the Samsung Tablets are more happy. And I definitely need the 4.1+ adapter as I want to re-broadcast my FTMS bike to Zwift and ANT+ at the same time.

closed 2021-02-01 by ptx2

chriselsen, 2021-02-02:
After testing this with a Raspberry Pi4, I can confirm that the Samsung Tab S4 now also sees power and cadence correctly via BLE.