ptx2/gymnasticon #32: [Peloton] Last cadence seen by Gymnasticon never clears if it is non-zero

jeremydk, 2021-01-26:
Reported to me over FB Messenger, trying to dig in.

We appear to have an edge case where if the bike is no longer sending updates (i.e. ride screen left), if the last frame contained a non-zero cadence, we emit pedal stroke events at that given rate until the service is restarted.

Thinking through this, we already have the means to account for this -- the PingInterval.  We currently reset the clock on the pingInterval with every PedalEvent, but we might be able to move the reset into the onBikeStats handler rather than onPedalEvent since they are being generated synthetically.

The other option is to try to detect the frames sent (if there is anything unique) when the serial handler on the Peloton Tablet is started and destroyed, and match that lifecycle within Gymnasticon, but it feels more fragile than adjusting our pingInterval logic.

jeremydk, 2021-01-26:
@ptx2 I'm happy to make these changes, but I don't have a way to test on non-Peloton hardware, and we don't have any real branching logic at that point in the application.

Before I go start crafting a PR, I'd love to make sure the intended fix is in line with your mental model of the situation as well.

ptx2, 2021-01-31:
Hey @jeremydk thanks for finding this! I'll try this on the Flywheel today.

commit 1dc4d0d7 (2021-02-02): Reset Peloton cadence/power when ride ends. (#37)

Fixes #32.
closed 2021-02-02 by ptx2