ptx2/gymnasticon #38: Can gymnasticon appear in Peloton's app as a cadence sensor?

gojomo, 2021-02-02:
Gymasticon is now working for me, via an RPi, for Zwift. Are any of its output modes capable of appearing as a cadence sensor to the Peloton app? (I'm not seeing it appear when Peloton App searches.)

ptx2, 2021-02-03:
Happy to hear it! I just pushed up #39 which adds the feature you described. To try it out, clone the repo:

```
git checkout -t origin/csc-server
npm install
npm run build
npm link
gymnasticon
```

gojomo, 2021-02-03:
Wow, thanks! I should be able to test it tomorrow.

lpecan, 2021-02-08:
Is there something you need to do to enable this? I updated to the latest version and did the Garmin permission trick. So ANT+ is working (which is great), but I don't see a BT CSC advertisement anywhere. I didn't see anything in the --help but thought maybe I'm missing something. I guess the concept is that (conceivably) in the latest update the rpi is outputting 1 CPS advertisement, 1 CSC advertisement, and an ANT+ cycling power/cadence message?

ptx2, 2021-02-11:
Added in #39 and will be in the next release.

closed 2021-02-11 by ptx2