ptx2/gymnasticon #42: Will this work on Keiser M3? Just need some pointer if possible?

nealjane, 2021-02-08:
I have a keiser M3 I- which has Bluetooth.

have previously setup ‘keiser2zwift at GitHub- but the app is incomplete with little help from the original developer for additional services for other cycling apps - peloton, RGT cycling etc.

Ive read this project with interest and I note the “ Support for other bikes can be added easily.” in the project description- ive read through the instructions on setting up with other bikes (numerous times!) - but for the life of me, apart from finding the relevant files and reading through them-  I’m not sure what I’m supposed to alter?!

Firstly- Will this app connect my bike over Bluetooth to a ‘pi zero W’, with an extra usb Bluetooth dongle to then forward the services to my apps?)
Secondly - I’m totally flumoxed to what needs adding to the files - eg ‘M3’ as the bluetooth title??

Any help available from anyone would be very grateful received!
Image attached showing Bluetooth details I get from the bike usingnRF connect..
[image]

chriselsen, 2021-02-09:
Here is a [BLE Simulator for the Keiser M3i](https://github.com/KeiserCorp/Keiser.M3i.BLE-HCI-Simulator), which includes documentation of the packet format. That should help with the development of a custom bike type.

nealjane, 2021-02-09:
Thanks @chriselsen - that all looks beyond me. 
Will keep an eye out and see if someone else comes along and knows how to do the technical work..!

chriselsen, 2021-02-14:
@nealjane Just created #50 for experimental Keiser M3 support. Give it a try and let me know how it went.

ptx2, 2021-02-21:
This has been released in 1.4.0.

closed 2021-02-21 by ptx2