ptx2/gymnasticon #46: CLI issues on iatest image (command not found)

barrettec777, 2021-02-10:
Hello!

I've deployed the image to my raspberry pi zero WH - and though gymnasticon is running, it seems like the CLI and or path settings may not be properly configured. Should it be working "out of the box"? When I type gymnasticon, I get 

```
pi@gymnasticon:~ $ gymnasticon
-bash: gymnasticon: command not found
```

I've been able to change settings (e.g. --bike "peloton") by modifying the gymnasticon.service file, but I'm wondering if there's something else that I should change to get the CLI to properly work?

chriselsen, 2021-02-10:
The executable for Gymnasticon lives within the folder `/opt/gymnasticon/node/bin/`, which is not part of the path, where the default shell searches for executables. The same applies also to `node `and `npm`, which are both installed in the same folder. 

Therefore you have to manually add this folder to the path via:
`export PATH=/opt/gymnasticon/node/bin/:$PATH`

This will only be of effect during the lifetime of the Shell / SSH session. If you want to make this change permanent, add it to the end of the `.profile` file within the user's home directory. 

E.g. via `echo "export PATH=/opt/gymnasticon/node/bin/:\$PATH" >> /home/pi/.profile` 

Don't forget to restart the Shell / SSH session after making this permanent change, for it to take effect.

barrettec777, 2021-02-10:
Great. Added that to my profile and can access the CLI now. I did have to run 
```
# this gives cap_net_raw+eip to all node programs not just gymnasticon
sudo setcap cap_net_raw+eip $(eval readlink -f $(which node))
```
In order to run as an unprivileged user (pi).

ptx2, 2021-02-12:
Nice catch @barrettec777. This will be the default in the next release thanks to @chriselsen.

closed 2021-02-12 by ptx2