ptx2/gymnasticon #48: Peloton Wiring Troubleshooting

jeremydk, 2021-02-11:
I think I might be having a similar issue. Here's my gymnasticon log (from my raspberry pi zero WH).

```
Feb 09 20:31:21 gymnasticon systemd[1]: Started Gymnasticon.
Feb 09 20:31:33 gymnasticon gymnasticon[269]: [2021-02-10T01:31:33.240Z] connecting to bike...
Feb 09 20:31:33 gymnasticon gymnasticon[269]: [2021-02-10T01:31:33.317Z] bike connected /dev/ttyUSB0
Feb 09 20:31:33 gymnasticon gymnasticon[269]: [2021-02-10T01:31:33.355Z] no ANT+ stick found
```
Gymnasticon connects just fine to Zwift on my phone/ipad. But I see nothing more in the log when I pedal, and no response in Zwift. When I disconnect from Zwift, I see this (which may very well be expected behavior):

`Feb 10 09:22:14 gymnasticon gymnasticon[269]: noble warning: unknown handle 64 disconnected!`

I have the same RCA cabling setup as shown in the first issue as well as in this one, but I use a UGREEN USB to MicroUSB adapter to connect the USB-Serial decoder to the pi zero. I have a regular Pi 4 coming in the mail in a couple days, so I'll give that a try as well. 

I do have a multimeter if there's something you'd like me to check. I did a quick connection check myself and it does appear that the RING and TIP are indeed swapped.

Edit: This was with a scenic ride started and a bit of pedaling.

_Originally posted by @barrettec777 in https://github.com/ptx2/gymnasticon/issues/21#issuecomment-776741009_

jeremydk, 2021-02-11:
Hi @barrettec777 --

Let's start with a few easy things:

1.  With a ride loaded (can be waiting at the Start / Bluetooth Audio / HRM page pre-ride), if you can run `cat /dev/ttyUSB0` and spin the pedals, can you confirm that you're seeing any data flowing (will look like garbage)?
2. The microUSB OTG adapter shouldn't impact things -- I've got a similar setup myself,
3. In my experience, this has been always due to either a missing connection, or having the serial Tx connected. -- If the serial Tx is connected, you'll usually stop seeing data show up on the Peloton screen itself.  Are you still seeing stats on the Peloton?

barrettec777, 2021-02-11:
Hey @jeremydk, thanks for your help!

1. Loaded a scenic ride, had it actually running. Spun pedals, but see no data flowing at all after running cat /dev/ttyUSB0. 
2. Sounds good.
3. No TX connected. Just RCA red (from female 3.5mm TRS to RCA adapter connected to bike) --> RCA coupler --> RCA white --> Serial USB adapter --> microUSB adapter --> Raspberry PI.

[image]

barrettec777, 2021-02-11:
So - I just figured I'd give the reverse a try in case the wires were switched up or something... 

Bike --> RCA white --> RCA coupler --> RCA red --> Serial USB adapter --> MicroUSB adapter --> PI. 

Now I'm seeing some data coming in. Example:
```
pi@gymnasticon:~ $ cat /dev/ttyUSB0
������������m����_{�������[���}�������������������������������{_��W��������������}
```
Peloton still receiving stats just fine. It seems as though this is more periodic/continuous - I'm not sure if spinning the pedals has any significant effect on the data / volume of data. Gymnasticon also isn't picking anything up in its log (via journalctl), still just the same as before:
```
Feb 10 19:17:20 gymnasticon systemd[1]: Started Gymnasticon.
Feb 10 19:17:32 gymnasticon gymnasticon[291]: [2021-02-11T00:17:32.596Z] connecting to bike...
Feb 10 19:17:32 gymnasticon gymnasticon[291]: [2021-02-11T00:17:32.780Z] bike connected /dev/ttyUSB0
Feb 10 19:17:32 gymnasticon gymnasticon[291]: [2021-02-11T00:17:32.821Z] no ANT+ stick found
```
So at least this is some potential confirmation that data is being sent from the USB decoder?

jeremydk, 2021-02-11:
Okay, no data coming in off the USB device at all means we can focus only on the cabling.  The path you described sounds correct -- I'm curious about the splitter at this point. Can I get a photo of the Y splitter itself? I'm curious if it's TRS or TRRS (a picture of the male plug sides would confirm that.) With the reversed direction, you might be seeing the "polling" from the tablet to the bike, which Gymnasticon itself will throw out.

barrettec777, 2021-02-11:
So going from Bike --> RCA Red --> RCA Red --> Serial produces the same result as the switched config above (consistent data regardless of pedaling). I don't see anything coming in on gymnasticon. 

Here's that config: 

[image]

I believe that my Y cable is just TRS. It's [this one from Amazon](https://www.amazon.com/gp/product/B0036DRYOS/ref=ppx_yo_dt_b_asin_title_o09_s01?ie=UTF8&psc=1). Should it be TRRS? 

Just for completeness sake, going from RCA White --> RCA White produces no data at all (so the same result as the first configuration as directed in your original post).

jeremydk, 2021-02-11:
TRS is good and correct -- I'm glad that we can dismiss that as a worry.

`cat` on the device will show data flowing constantly -- the cadence/power information is both requested and sent many times a second, so that's not a surprise.

We can infer that serial "red" is your Rx pin since we are seeing data flowing. We should be able to confirm that with the product listing as well.

For kicks, can you flip the connections on the splitter?  From what I'm seeing, things should be okay. I might need to throw together a debug/raw mode build to help troubleshoot things like this in the future

The full chain is that:

Tablet sends REQUEST frame to Bike. Bike sends CADENCE or POWER frame to Tablet, we snoop the Bike Tx via serial RX + splitter, and decode the frames. We shouldn't see the tablet REQ frames because they'd be on the other wire.

Lemme pull a PR together to add some deeper logging from a gymnasticon perspective as we continue to look at this.

barrettec777, 2021-02-11:
Awesome - thanks for your help, man. I switched the connections on the splitter, same result. 

Also - I tried wiring this up:

[image]

I think this accomplishes what we want, no? Does that look correct? Maybe this'll help with the troubleshooting. 

When connected such that **Bike** --> **RS** --> **TS** --> **USB Adapter**, I see a significant increase in data compared to before (regardless of pedaling), but nothing from gymnasticon. When I turn my cable around (Bike --> TS --> RS --> USB Adapter), I see no data.

jeremydk, 2021-02-11:
That should help as well, since we can actually tightly control the connections and iterate through.

I've got a PR up at https://github.com/ptx2/gymnasticon/pull/49 -- setting the NODE_DEBUG environment flag to "gymnasticon:bikes:peloton:trace" with that build will start to log the raw messages from the bike or tablet.

barrettec777, 2021-02-11:
Awesome. Sounds good - will try when the pr gets through.

Harti-801, 2021-02-15:
Hi, I am really new here. I tried this nice project out but it seems I have a problem with the cabling.
The USB serial adapter recommended is not available in Germany. I am not able to find this kind of adapter that includes the MAX3232 chip. Am I right that this one is needed? Maybe someone has a recommendation what is working. I allready tried with an adapter from DIGITUS (USB 2.0 SERIAL ADAPTER). The Pi stops here: bike connected /dev/ttyUSB0

jeremydk, 2021-02-15:
Most USB serial devices should be fine -- the biggest question is if they are Tx/Rx Tip/Ring or Ring/Tip -- you need to ensure that the receive pin of the serial is connected to the transmit pin of the Bike (not Tablet).  I can offer advice, from afar, but this is hard to walk folks through due to their own personal setups.

jeremydk, 2021-02-16:
Drew this up, this should help folks get things wired successfully. 

[image]

jeremydk, 2021-02-17:
Built a little switch box per the diagram above today and can validate that it works with the recommended USB serial device.  For folks having trouble, please validate the Rx/Tx pinout of your USB Serial device, and ensure your wiring matches the diagram above.

ptx2, 2021-02-21:
@Harti-801 @barrettec777 Are you still having problems with the cables?

Harti-801, 2021-02-22:
thanks to the diagram I could double check my cabling, I found no mistakes here.
But as I am not sure with my adapter, I tried to organize the recommended one. I hope to get it end of this week.

chaloney, 2021-03-01:
i bought the recommended cables but the rca connectors from monoprice are still in transit so i cut and resoldered the serial cable, reversing the tip and ring wiring (and shortening the cable to a reasonable length). i have both tx and rx connected and no connection to the peloton tablet to use the direct polling from gymnasticon.  gymnasticon connects ok to trainerroad on my laptop but there power and cadence stay at 0 when pedaling.  i went back and rechecked my soldering with a multimeter and it turns out the rx was initially connected to the ring so i cut the wires and resoldered back the way they came with the white wire connected to the ring, the green connected to the tip, and the black to the sleeve.  i still get no data from gymnasticon.  i have another 3.5mm coupler coming, although the one im using has worked in other uses.  i also have a replacement usb to micro usb adapter en route.  i flashed the most recent gymnasticon version (as of 2/25/2021).  does the direct polling function automatically or did i miss a step?  i dont have the right splitter right now (or the rca cables / coupler to disconnect the tx side) to check.

i also found a different serial decoder that claims to have the opposite rx/tx wiring, so no messing with cables should be necessary (although that should be the case with the one i received also):
https://www.amazon.com/gp/product/B08FDJS58X/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1

[image]

jeremydk, 2021-03-01:
Please try turning on the trace logging, as well as giving us the output of ‘lsusb’ and ‘ ls -al /dev/*USB*’

It isn’t feasible for to begin to troubleshoot without understanding if this is a physical wiring issue or a software one.

Direct polling is a default behavior, so this is either a wiring issue or a setup issue with the USB serial.

chaloney, 2021-03-08:
unfortunately my raspberry pi skills were limited to creating an XMBC server about 7 years ago so I need more guidance than that to give you useful information.  i have tried every combination of connections possible with the RCA cords/coupler both with a "just ride" running and without, with the serial decoder i mentioned above and with a new, unaltered one.  power and cadence remain at 0/0 (using trainerroad as that is why i am doing this).  

i know i saw a mention of trace logging somewhere on the github but i cannot find it now.  
[image]

this is while the pi is connected to the bike, both tx (tip) and rx (ring)

jeremydk, 2021-03-08:
It looks like you purchased a USB to TTL adapter,  not an RS232 as linked.
The voltages (among other things) differ (3.3v vs 5), and it is unable to
read the signals from the Peloton bike.

Was helping someone else via Facebook where we discovered they made the
incorrect purchase as well.

chaloney, 2021-03-08:
the first purchase was from the amazon link kevtainer posted in the peloton bike decoding pull request, obviously the seller changed the item.  it looks like the link is in stock again with the correct ("not ttl" description) will try again.  thanks.  the second decoder i bought from the link in my first post is the same pcb with the same chip as the first one.  hopefully better luck this time.  thanks for your help

jeremydk, 2021-03-08:
You’re specifically looking for rs232, and as long as it says that you
should be good to go.

chaloney, 2021-03-08:
the original link now specifically says rs232, "not TTL"  i dont know if it said that the first time i ordered but whatever it said, they sent me the wrong thing.  now i know what to look for.  thanks.

chaloney, 2021-03-12:
New cable works! Thank you!!

jeremydk, 2021-03-12:
Fantastic -- glad that worked out!

tanzmanl, 2021-05-06:
Hi Everyone,
I'm seeing the same issue as the original post. Connecting to bike, bike connected /dev/ttyUSB0, no ant+ stick. My Zwift app shows Gymnasticon, but the cadence/power just stick at 0 as I pedal and the logs in journalctl don't seem to add anything further. When running cat /dev/ttyUSB0, I do get a bunch of garbage data coming through. 

How do I enable the NODE_DEBUG variable? Is that just export NODE_DEBUG="gymnasticon:bikes:peloton:trace" and then just restart Gymnasticon? Where I do see the logging from that and what other helpful logs can I provide? Thanks!

[image]

kschick, 2022-04-08:
I'm not sure if anyone is still paying attention to this but I had a question - I have this working currently on Zwift and with peloton as well but I need to unplug one of the RCA cables for it to work on peloton. This isn't really a problem but I just want to get a better understanding. My hardware understanding is pretty minimal so I'm not sure if I can describe this correctly. Basically, I think I kinda get why it doesn't work with peloton as the data going back on the receiver cable is messing things up and that I've seen mentioned in a couple places on here. 

What I don't understand is why it doesn't output power/cadence without both cables plugged in. If I only hook up red to white as is shown by most of the images here I see 0w and 0 cadence. I tried ever other combination with only one cable and its the same thing. The only thing that works is red to white and white to red both plugged in. I don't understand this as, with my basic understanding, the only thing that matters is that peloton is transmitting to the pi and anything going back to peloton shouldn't matter and in fact causes problems when using the peloton software. Can anyone shed some light on this?

gbak, 2026-09-13:
Anyone still interested connecting to a Peloton.  Aside from flashing the image to the PI, it takes all of 10 minutes to connect everything, worked first try.  Awesome project

- ANT+ - https://www.amazon.com/dp/B087QB1Q7M
- USB RS232 Pigtail - https://www.amazon.com/dp/B08BNLH1FB
- TRC Splitter - https://www.amazon.com/dp/B000067RC4
- TRC Male Pigtail - https://www.amazon.com/dp/B08133YLZ7

```
RS232       TRC Pigtail
Black   <->     Black
Yellow  <->     White
```

doudar, 2026-09-13:
It's just a little outdated since we can do this on device with no additional hardware using Grupetto nowadays.