ptx2/gymnasticon #51: mpaceline/ismoothrun support?

tompijls, 2021-02-16:
Hi

trying to use gymnasticon with mpaceline and ismoothrun.
Just did a clean install from image on a RPI 4. I intend to use it with the Keiser M3i bike, for which PR 50 is being evaluated. 

As these apps/bike are not officially mentioned as compatible, i decided to start with the bot:
gymnasticon --bike bot --bot-power 150 --bot-cadence 80

This seems to report accurate values in both apps.

I do get however frequently below error
noble warning: unknown handle 64 disconnected!

Any idea why this happens?

lpecan, 2021-02-17:
No idea why it's throwing the error for you. Since ANT+ became available, I did test mpaceline with gymnasticon where I connected mpaceline to gymnasticon over BT CPS and an android bike computer to gymnasticon over ANT+. It worked perfectly. The one thing I couldn't get to work (not that I really care) is that once connected to the BT CPS on mpaceline, there was no way to connect separately to the BT CSC advertisement. Seems like it is either or.

chriselsen, 2021-02-20:
This warning seems to be coming from [here](https://github.com/noble/noble/blob/c4cd18a7a429bb832f6c4ca793be3faf9af884e9/lib/hci-socket/bindings.js#L261) within the noble library. 
It could be related to this [other reported issue](https://github.com/ptx2/gymnasticon/issues/55) and probably just indicates that your device running mpaceline or ismoothrun temporarily lost connection.

tompijls, 2021-02-21:
@ptx2 you can add mpaceline support on the description. it works properly now for me

closed 2021-02-21 by tompijls