ptx2/gymnasticon #53: Strava export support?

tompijls, 2021-02-19:
Maybe slightly out of scope for gymnasticon, but anyone ever looked into building export capabilities to eg strava?
Would mean you can use whatever app you like, and still export those stats to strava (or any other integration which gets built)

chriselsen, 2021-02-20:
Not sure if that makes sense to have it built into Gymnasticon itself: I see Gymnasticon - especially with the Raspberry support - as an embedded device that doesn't need to be connected to over Ethernet/Wifi or powered on/off. Instead it's a bike add-on without interface.  
Exporting data from that device would require storing data (wears out the SD-card), an export mechanism and network connectivity. 

A better approach would probably an Android / iPhone app that can pick up the Gymnasticon sensor data by BLE or ANT+ and then upload them to Strava. 
That's e.g. a main driver for the ANT+ support: Record the rides via Garmin devices.

ptx2, 2021-02-21:
It is outside the scope but...

It wouldn't be too hard to record a FIT file for each ride.

* Automatically start recording on the first stats packet
* Automatically stop recording some time after the rider stops producing stats
* Save the recording to disk and queue it for upload
* Strava endpoint and key could be provided in /boot
* Any additional relevant data (rider weight) could also be provided in /boot
* Ride files could also be grabbed manually from http://gymnasticon.local/
* Add heart rate sensor support so that data can be included in the recording
* Old rides can be removed to make space for new ones to ensure disk doesn't fill up
* If you have a read-only filesystem, the rides just disappear at each boot
* Automatic mode could be optional and a simple web UI could offer manual start/stop/upload

I agree with chriselsen that this should probably be a separate program. It could still live on the Pi and receive data from Gymnasticon directly. The changes required to Gymnasticon itself could be minimal: optionally receive/rebroadcast heart rate and publish all stats over a socket.

Not sure if this would address your need @tompijls?

It's worth pointing out that wherever an app has Strava integration the data will probably be richer. e.g. Zwift can include elevation data that we won't know about.

If it can be kept simple and still be useful to enough people I'm open to the idea.

tompijls, 2021-02-21:
Thanks both - I don't necessarily disagree with you , but then again, gymnasticon solves the problem of connecting older/poorly implemented equipment to modern apps.

Extending strava (or finalsurge/trainingpeaks) connectivity for apps who don't support these integrations natively does kinda fit as well ;-)

I'll see if i can give it a go in the coming weeks (in my case, it is because mpaceline does not allow for strava integration)

Cheers

nealjane, 2021-02-21:
> Maybe slightly out of scope for gymnasticon, but anyone ever looked into building export capabilities to eg strava?
> Would mean you can use whatever app you like, and still export those stats to strava (or any other integration which gets built)

What’s wrong with just connecting keiser to ‘keiser m series’ app for sessions - it syncs with twin peaks and Strava??

tompijls, 2021-02-21:
HA, i was gonna say laziness not to have to start 2 apps at the same time... then again, it's probably gonna take a lot more time to get the integration built as I'm not really familiar creating fit files.

low priority for sure