ptx2/gymnasticon #61: Fix file-not-found issue in pi-sdcard build process

ptx2, 2021-02-23:
Running `./build.sh` eventually results in this error:

```
install: cannot stat 'files/gymnasticon.json': No such file or directory
```

It can be worked around by continuing the build:

```
cd pi-gen
CONTINUE=1 ./build-docker.sh
```

But it would be great to fix it so it runs all the way through on its own.

chriselsen, 2021-02-28:
I added a `pwd` [here](https://github.com/ptx2/gymnasticon/blob/75caf7619ea05e488bcdbac859200c08f30f0802/deploy/pi-sdcard/stage-gymnasticon/00-install-gymnasticon/01-run.sh#L27), to see what path the `01-run.sh` script is in during the first failing execution and then during the next round. 

First time, when the build fails the path is:
`/tmp/tmp.7fP5TlDQV5`

Second time, when the build succeeds, the path is:
`/pi-gen/stage-gymnasticon/00-install-gymnasticon`

Looks [like ](https://github.com/ptx2/gymnasticon/blob/75caf7619ea05e488bcdbac859200c08f30f0802/deploy/pi-sdcard/stage-gymnasticon/00-install-gymnasticon/01-run.sh#L9-L11)this is the reason for moving into the /tmp dir.

We can fix this, by executing `STAGEDIR=$(pwd)` before the if block in line 7 and `cd $STAGEDIR` after the if block in line 23.

chriselsen, 2021-02-28:
Ended up seeing the cause for the `cd $TMPD` and fixed that instead. See pull request #69.

ptx2, 2021-03-06:
Fixed by #69.

closed 2021-03-06 by ptx2