ptx2/gymnasticon #64: Error extracting archive.  Lzma library error

anijohns, 2021-02-24:
I have a flywheel and recently purchased a Raspberry PI.  I tried to follow previous instructions and spent hours setting up the PI.  Today, I learned that all I had to do is insert the SD card into my computer and add the Gymnasticon imager using the Raspberry Imager.  However, I received this error (Error extracting archive.  Lzma library error.  No progress is possible.  Please help as I am not a tech person but since I have spent time on this, I would like to get this working.  I have the QDZ app but cannot get my watts to show up in Fulgaz.  Hoping this will fix the issue.

ptx2, 2021-02-24:
Hi!

It sounds like maybe the image file didn’t finish downloading.

1. Can you include a screenshot of the error?

2. What is the exact filesize in bytes (not kB, MB or GB) of the
Gymnasticon image file (gymnasticon-raspberrypi.img.xz) on your computer?

3. What version of the Raspberry Pi imager and what OS?

The above information will help narrow it down.

Things to try in the meantime:
* Download the file again
* Try decompressing the image file (.img.xz) with another program e.g.
7-zip. Then try Raspberry Pi Imager with the decompressed (.img) file.
* Try the downloading/imaging steps on another computer if you have one
* Try a different SD card if you have one (likely not the problem)

anijohns, 2021-02-24:
I will try again.  Just didn’t think this would be so difficult.  Here is the snapshot of the error.  It wouldn’t let me proceed.
 I have Rasberry Zero and downloaded on a PC. I may try the MacBook Air tonight to see if that works.

Kind Regards,
Anita -

anijohns, 2021-02-25:
Hi, you guys are genius!!

I tried another computer with Windows 10 and it worked beautifully.  Trying to connect my bike in Fulgaz now and have been waiting for the Trainer Resistance Control to connect?  

See pix...should I just not use this?

Kind Regards,
Anita -

ptx2, 2021-02-25:
It sounds like the Raspberry Pi is having trouble finding your Flywheel
bike.

First things to check:

   1. Make sure you are pedaling while you are trying to connect. The bike
   only appears when pedaling. It will disappear 1 minute after pedaling stops.
   2. Make sure the Raspberry Pi is near the bike. Same room should be fine
   generally. But try moving it close as possible for now and removing any
   obstacles between the two.
   3. Make sure the bike has fresh batteries.
   4. Only one device can connect to the Flywheel bike at a time. Make sure
   no other apps or devices are connected to your Flywheel bike. Go into
   Bluetooth settings on each of your devices (phone, tablet, pc) and if you
   see “Flywheel 1” disconnect it. You could also just shut those devices off
   while troubleshooting this to be sure.

There is further troubleshooting that can be done but those are the most
common issues and a good place to start.

anijohns, 2021-02-27:
OK I tried below but it is still trying to connect the Trainer Resistance Control.  Funny because the power, cadence and speed connects fine.  Any other tips?

Kind Regards,
Anita -

anijohns, 2021-03-04:
Hi, I am following up.  I did see that you all tested the Fulgaz app.  Did anyone have trouble connecting the trainer resistance control?  As I mentioned, the cadence, speed, and power all works and shows well on the display.  
I do have the QDZ (qdomyos) app on my phone, but had it turned off so it is not the culprit.
Please advise if you have any other suggestions to get it working.
Thank you for your help.  Anita -

Anita C. Healey, Principal

Custom Pro Files LLC

www.anitahealey.com

www.linkedin.com/in/customprofiles

Phone/Fax: +1.888.515.4911

ptx2, 2021-03-06:
Hi @anijohns

Happy to hear you got power and cadence working!

The Flywheel bike does not support electronic resistance control which is why the trainer control feature is not working. You can still use apps like Zwift and FulGaz but will need to change the resistance by hand with the dial. This is a hardware limitation of the Flywheel bike.

In theory, you could add electronic resistance control to the bike by adding a motor to change the resistance. The links below may be of interest in that case:

https://github.com/ptx2/gymnasticon/issues/26
https://github.com/doudar/SmartSpin2k

anijohns, 2021-03-06:
Thank You!

Kind Regards,
Anita -

closed 2021-03-06 by ptx2