ptx2/gymnasticon #65: Adding gymnasticon.json to boot makes 2 gymnasticon services run at startup

nealjane, 2021-02-25:
I know this is a work in progress as you’re both adding stuff on a daily basis and it’s not a big issue- more of a comment.
Current image 1.4 with keiser added written to sd card and gymnasticon.json added for keiser bike before first boot makes 2 gymnasticon services run on boot 
(I’m assuming that’s the case - SSH login it’s shows the gymnasticon logo twice on the screen.)
Thanks for all the work on this to you both @ptx2 @chriselsen it’s working great.

chriselsen, 2021-02-25:
There are actually two gymnasticon services that get started on boot:
- [gymnasticon-mods](https://github.com/ptx2/gymnasticon/blob/main/deploy/pi-sdcard/stage-gymnasticon/00-install-gymnasticon/files/gymnasticon-mods.service): That service pulls in the JSON config file from /boot to /etc. It should terminate very shortly after boot. 
- [gymnasticon](https://github.com/ptx2/gymnasticon/blob/main/deploy/pi-sdcard/stage-gymnasticon/00-install-gymnasticon/files/gymnasticon.service): The actual service that runs the "gymnasticon" binary.

After login you can check with `ps ax` if there is more than 1x gymnasticon process running. 

Will need on check the 1.4 image to see about the SSH login. The gymnasticon icon comes from the /etc/motd file and I haven't seen any changed to that file.

nealjane, 2021-02-25:
Ah! I’ve only tried/seen it first boot after adding the json to boot- just rebooted the pi and it’s now 1 logo after login.

closed 2021-02-25 by nealjane