ptx2/gymnasticon #67: Feature request: Expose HR

tompijls, 2021-02-26:
Some bikes (eg Keiser M3i) retrieve and broadcast HR. Some older equipment maybe also does this through HR sensors in handle. Could be an idea to optionally expose this as a HRM Profile

chriselsen, 2021-02-27:
For heart rate the serviceUUID is 180d and the characteristicUUID is 2a37. XML file for BLE description is [here](https://www.bluetooth.com/wp-content/uploads/Sitecore-Media-Library/Gatt/Xml/Characteristics/org.bluetooth.characteristic.heart_rate_measurement.xml).

This will also help for the case where Zwift or other apps run on an Apple TV, as the Apple TV can only talk to 2 devices over Bluetooth at a time (in addition to the TV remote control.

crissag, 2021-03-05:
I would love to see this implemented as well, unfortunately beyond my current time constraints. I have a Keiser M3i and wear a Polar H10 HRM and my heart gets sent to my Apple TV or iPad/iPhone via the Keiser M Series convertor. Unfortunately, the Keiser convertor just doesn't stay connected consistently like Gymnasticon does.

chriselsen, 2021-03-06:
@crissag The Polar H10 HRM actually supports Bluetooth and ANT+ and you could therefore connect it directly to the Apple TV. That's what I do with a Garmin heart rate strap and it proves to be more reliable that having a bike as man-in-the-middle. 
In my case this adds up to Gymnaticon + HR strap + Apple TV remote = 3x bluetooth devices and therefore comes in within the limit.
If I would need to add a bluetooth based speaker or headset at this point I could not. 

Just keep in mind that most Bluetooth sensors like the Polar H10 HRM can only be connected to from one device at a time. Therefore if you don't see it from e.g. Zwift on Apple TV, that connection has probably already been "grabbed" by your bike.

nealjane, 2021-03-06:
Alternatively for Apple TV you can use phone as the zwift receiver (to connect to all Bluetooth accessories eg bike and hr) and then sends to Apple TV - which then leaves you with 1 free bluetooth remaining for headphones....

crissag, 2021-03-06:
Before Gymnasticon, I used my Polar H10 with a Polar watch and then the Keiser to Apple TV/Peloton app through the M Series Convertor. The Peloton app is capable of displaying HR on screen (newly added feature). I eventually lose the HR signal on the Peloton app because the M Series Convertor has been very flaky. My desire in getting this feature added would be for the HR to go through the Keiser bike to the Peloton Apple TV app.

A nice feature with the Polar H10 is that it actually supports two Bluetooth connections (configurable via Polar Beats app). 
[image]

nealjane, 2021-03-06:
It is doable without going through keiser bike though..but can see why you might want it...because the ‘keiser converter’ does it...

I’m just grateful for what Chris and ptx2 have done do date to be honest - it’s been working great. I only mentioned adding the bike a month-6 weeks ago... 👏👏

crissag, 2021-03-06:
Agreed, very grateful for the work of everyone here. I'm only requesting this feature because I the iOS versions of the Peloton app are receiving cadence, so I imagine that will be added to the Apple TV app as well in the future (they recently added heart rate and have been enhancing their Apple TV app more than other streaming platforms). Hopefully, someday they will also add power/watts display as well. This would be a brilliant solution to provide that consistent connection and all the details that are helpful to have onscreen.

nealjane, 2021-03-27:
@crissag Djwasser has managed to get HR working partly on his M3i pi app here- 
https://github.com/djwasser/M3iPiZeroW