ptx2/gymnasticon #68: Feature Request: add speed to M3i bike

tompijls, 2021-02-26:
M3i does not broadcast speed, however does expose total distance and cadence, so should not be too complicate to derive speed and expose this in the CSC profile.

chriselsen, 2021-02-27:
What usage are you hoping to get out of speed data from the Keiser M3?

Apps like Zwift do not use speed sensor information directly. Instead the [in-game speed is calculated](https://zwiftinsider.com/how-does-zwift-calculate-my-speed/) based on power data, received from the bike. This in-game speed greatly depends on e.g. the gradient of the road within the game. Therefore when you output 200W from your bike, this could translate to a speed of 40 km/h on a flat course, or to just 5 km/h on a course with a 10% gradient. 

For bikes like the Schwinn IC4/IC8 or the Flywheel which do show speed information on the display and transmit them over Bluetooth, this speed information usually does not match what you see in-game. The bike effectively assumes an all-flat course. As a result the Internet is [littered with questions and complaints ](https://zwiftinsider.com/garmin-zwift-distance/ )about why these numbers don't match. 
Therefore while almost trivial to pass the speed information received from Schwinn IC4/IC8 or the Flywheel on via BLE CSC information, the usefulness of that data is questionable. 

In the case of the Keiser M3 there is another challenge: As you already wrote, the bike doesn't transmit current speed information. Calculating the current speed based on distance traveled and time elapsed is not possible. This will only give you average speed over the time elapsed. Imagine you bike 20 km, which takes you 60 minutes. While that tells you that your average speed during that ride was 20 km/h, it doesn't mean at all that at the 60 minutes mark you were actually riding at 20 km/h. For BLE CSC we need current speed. 

There is a relation between cadence, resistance (in case of the Keiser M3 transmitted as "Gear") and speed, which can e.g. be seen [here](https://www.bikecalc.com/speed_at_cadence). As the Keiser M3 is a spin bike and not a regular bike the gears don't match any such tables and instead you would have to create one yourself. 
But how? You don't have a speed value displayed on the bike's built-in computer to determine what speed would match 100 rpm @ Gear 1 and therefore what to put into that box.
Instead you would have to bike for e.g. 10 minutes at exactly 100 rpm and Gear 1 and then determine based on the shown distance the average speed for that time interval. That's what would goes into the 100 rpm @ Gear 1 box. Now repeat the same for all Gears and reasonable cadences between 30 and 130 rpm. There a lot of placed to introduce inaccuracies in that process, so the calculated speed would still only be a very rough guess. 
That's a lot of work to get a value displayed that has a very questionable meaning.

tompijls, 2021-02-27:
Hey @chriselsen 

i don't use zwift, so it was more a general request. I do understand relationship between gear and speed, and average vs instant speed ;-)

On the keiser, the gear is more a break. One rotation of the pedals, equals to about 8.5 rotations of the flywheel. So there is a 1-1 relationship between cadence and speed actually. Whether it corresponds to something in real life on the other hand is questionable - i fully agree.

closed 2021-03-06 by tompijls

chriselsen, 2021-03-06:
For bike types that deliver speed information to Gymnasticon (e.g. IC4 or Flywheel), we should be able to easily pass this through via BLE and ANT+ to receivers.