ptx2/gymnasticon #70: Code cleanup: Deprecate --flywheel-address in favor of --bike-address

chriselsen, 2021-02-28:
Currently only the flywheel bike driver supports "locking" the bike to a certain Bluetooth MAC address via the `--flywheel-address` option. It would make sense to expand this capability to the other two Bluetooth enabled bike types (ic4, keiser) and as a result rename the option to `--bike-address` or `--bike-mac-address`.

There shouldn't be a need to do the same with `--flywheel-name `, as the LocalName for ic4 and keiser bikes is fixed. But with the flywheel bike, using the  `--flywheel-name ` option could be used to make gymnasticon work with a LifeFitness IC5.

goosebumps827, 2022-10-27:
Would you mind if i handled this for you guys?

nealjane, 2022-11-09:
Hi Goosebumps - I think any help is much appreciated by us gymnastic users.