ptx2/gymnasticon #71: Polar (Vantage V) Support?

Harti-801, 2021-02-28:
I did a first ride with my Peloton connected to a Polar Vantage V. So far this device was not mentioned as supported or rather tested device.
During my first test ride with the new setup two issues occured:
1. On the Polar the data is not showing in real time. So there is a significant delay of cadence and power. I guess that the gymnasticon sends to much messages so that the Polar BT receiver is not able to handle this.
2. the cadence dropped very often to a value of about 44 - 46, very seldom to values of about 30. Probably a side effect of point 1.

I could not find any issue like this, so I add this new one. Did anyone has experiences with a Polar Vantage V?
What I will do next is to double check with a Zwift setup or any kinde of BT app to make sure that my setup is working like expacted.

chriselsen, 2021-02-28:
Gymnasticon behaves like a regular cadence and power sensor, where data is based on the simulated crank revolution. There isn't really such a thing as "sending too much messages". 
But some watches or bike computers will not show the realtime values for cadence or power, but rather 3 second averages. It's possible that's what you are seeing.  

To make sure the communication between gymnasticon and watch is working, try to run gymnasticon in "bot" mode, e.g. with this config file:
```
{
  "bike": "bot",
  "bot-power": 200,
  "bot-cadence": 95
}
```

That will make gymnasticon output a constant cadence of 95 rpm and power of 200 W. You can then check on your watch what you see and make sure it matches.

Harti-801, 2021-03-01:
Understood.
I did some short tests with the bot:
- Some minutes with Zwift without any cadence drops
- Some minutes with Polar with cadence drops down to something around 45 than back to correct value

I was not able to test the delay of cadence and power with the bot.

Upper diagram is power with two drops where I stopped the bot. Lower diagram is cadence with max. 95. The measure is 6 min 45 second long.
[image]

chriselsen, 2021-03-01:
These drops indicate a Bluetooth connection issue between the device running Gymnasticon and your Polar watch. What device are you running Gymnasticon on? 
If you're using a Raspberry 3, that device is known for having Bluetooth issues, when using Wifi at the same time. You might want to disable Wifi in that case. Also moving the device running Gymasticon closer to the watch might help. 

Also check if the Polar watch supports ANT+. And alternative would be to use Gymnasticon with ANT+ for the watch.

Harti-801, 2021-03-01:
Gymnasticon is running on Pi Zero W. During the test the watch was less than 40 cm away. But WiFi is active, I will make tests with disabled WiFi.

chriselsen, 2021-03-01:
The Pi Zero W is OK, when it comes to Bluetooth + Wifi. I also use that platform without any issue. But I use it with a Garmin watch and an external ANT+ stick. Although that's working from 2-3 m away without any issues. 

Someone with more knowledge of the Bluetooth protocol might be able to look at log files and see what the issue with the Bluetooth dropouts is and if it can be fixed from Gymnasticon side.

Can you collect some Bluetooth logs from Gymnasticon:
```
btmon -w polar-ble.log &
systemctl stop bluetooth
systemctl start bluetooth
gymnasticon --bike bot --bot-power 200 --bot-cadence 95
kill %1
```

Once you see a few dropouts on your watch, you can kill the gymnasticon process and also btmon.

ptx2, 2021-03-06:
Hi @Harti-801 

> I was not able to test the delay of cadence and power with the bot.

It is possible to change the bot's power and cadence at runtime by sending it a JSON object over UDP. e.g.

```
echo '{"cadence": 101, "power": 202}' |nc -uw0 127.0.0.1 3000
```

Also, are you able to provide the raw data that graph is based on? e.g. a csv of "timestamp, power, cadence" ?

ptx2, 2021-03-07:
@Harti-801 Btw, if you run the bot with cadence=60 do you still get dropouts?

Harti-801, 2021-03-10:
@ptx2 attached a csv file that contains the data of the graph.

Harti-801, 2021-03-10:
I created a log as recommended by @chriselsen 
I have tested with cadence=60. In this case the cadence drops to 30. The power is stable as usual.

Unfortunately I was not able to change the values over the UDP command. Maybe I make typos or something totaly wrong. I will try again.

[Christian_H_2021-03-10_08-42-52_polar-ble_3.ZIP](https://github.com/ptx2/gymnasticon/files/6114174/Christian_H_2021-03-10_08-42-52_polar-ble_3.ZIP)
[polar-ble_3.log](https://github.com/ptx2/gymnasticon/files/6114175/polar-ble_3.log)

ptx2, 2021-03-10:
> Unfortunately I was not able to change the values over the UDP command. Maybe I make typos or something totaly wrong. I will try again.

Sorry, try adding `--bot-host 127.0.0.1 --bot-port 3000` to the gymnasticon command-line. (In the main branch and in future releases it's not necessary, but in the current v1.4.0 and earlier releases it is).

Thanks for attaching the logs!

ptx2, 2021-04-21:
Hi @Harti-801 are you still having issues with this?

Harti-801, 2021-04-27:
Hi @ptx2, sorry for no updates that long time. Yes, the issue is still the same.
I have made a test of 3 hours. The delay grows approximately 1s per minute.
Further I tried the newest version without enabling Wifi, but the cadence drops are still the same. A real test concerning delay is still pending.
I like to make a test without sending cadence. I like to see if the delay is still there. But unfortunately I did not find the time to find out how to make changes.

I am wondering if I am the only one using a Polar device.