ptx2/gymnasticon #75: Output speed/cadence on ANT+ too for IC4

djfjeff, 2021-03-19:
On my IC4 bike, gymnasticon seems to grab the power, speed and cadence BLE data from my bike successfully. It also seems to output my power data on my Garmin ANT+ stick (connected to gymnasticon Rpi) correctly.

However, it seems it output my speed and cadence over BLE only, I see nothing for speed and cadence over ANT+.

I was under the impression that it would also output the speed and cadence over ANT+ too so I could use it elsewhere.

Is that a misconfiguration on my end or is it the expected behaviour?

djfjeff, 2021-03-19:
Also, it seems that while my bike output speed over BLE, gymnasticon take a hold of the BLE connection but does not retransmit speed at all (either over BLE or ANT+). So my speed data is not available anymore.

So ideally, it would look like this for the IC4 bike :

Gymnasticon grab the power, speed and cadence from the bike over BLE
Gymnasticon then output :
 - Power, speed and cadence over BLE
 - Power, speed and cadence over ANT+ if a stick is present

This way, the data could be used by multiple endpoint (watch, laptop and bike computer).

Thanks,

ptx2, 2021-03-20:
Hi!

Re: cadence not working.

Currently we output:

* ANT+ Bike Power Profile (power and cadence)
* BLE Cycling Power Service (power and cadence (as crank revolutions))
* BLE Cycling Speed and Cadence Service (cadence-only; intended for use with the Peloton app)

In theory, you should get cadence if you're getting power. Can you share details on your setup so we can investigate further? e.g. which app/device, versions, OS version, bike firmware versions, gymnasticon version, and any screenshots and logs or other info you think relevant.

Re: outputting speed.

The reason Gymnasticon does not output speed is because in most cases the app (e.g. Zwift) knows more (hill grade, weather, etc.) and can calculate a more accurate value. We could pretty easily pass it along if there is a need for it, though. Can you share a little more how you're using the speed value?

djfjeff, 2021-03-20:
Hi,

The main reason for speed data is simple : I don't use Zwift or other apps right now, so I have no way of getting this data on my watch any other way.

I know this is not a perfect metric (power is not too as every spin bike "estimate" this number too) but I like having it to compare between run.

Also, my IC4 bike console display started to do weird stuff so I guess it might go out someday.

The reason why I also want everything broadcasted on ANT+ is I would like to link multiple display in the future (my watch and maybe a laptop screen to replace my bike console display).

Thanks for your work!

ptx2, 2021-03-20:
Ah ok, that makes sense. I think it shouldn't be too hard to capture speed and pass it along.

You also mentioned you were having problems getting cadence data via ANT+? Both power and cadence are broadcast over ANT+ in the Bike Power Profile. Can you share more details? e.g. which device are you using, does it just show 0 cadence or something else, etc.?

djfjeff, 2021-03-20:
Thanks for taking the time to check for speed data.

For cadence, let me double check, it might be an error I made on my end. On my Garmin watch, the sensor is called Power so I assumed it was only power data, so I also linked the BLE cadence sensor to get cadence.

I will try with only the Power sensor to see if it will provide the cadence also.

djfjeff, 2021-03-22:
Ok, you were right about the Power/Cadence stuff. I did a test and linking only the Power ANT+ sensor provide BOTH power and cadence, so no need to also link the Cadence BLE sensor.

So I would only miss the speed broadcasted on ANT+ to complete my setup!

djfjeff, 2021-03-22:
I also think it would be great to broadcast a "normal" Speed/Cadence sensor over ANT+ also (in addition to the Bike Power profile) because many watches from Garmin does not support power profile and only support the "normal" Speed/Cadence sensor (namely the Vivoactive 3/4, Venu, low budget Forerunner, etc).

ptx2, 2021-03-27:
Some thoughts:

  * The IC4 gives us instantaneous speed (km/h) in the FTMS Indoor Bike Data characteristic.
  * The IC4 also offers a CSC service containing wheel revolution data. See https://github.com/ptx2/gymnasticon/issues/34#issuecomment-770443252 for a log.
  * BLE CSC and ANT+ SPD/S&C want speed as: cumulative wheel revolutions + last wheel event timestamp.
  * One option (1) is to convert that km/h value to wheel revolutions.
    * Simulation() already handles this same type of conversion for cadence (rpm to crank rev). It could be reused here, with a little renaming.
    * Wheel circumference needs to be known for that conversion and can be provided via CLI/config. If there's a default value used by most receiving devices, we should make that the default here too.
  * Another option (2) is to subscribe to the IC4 CSC service.
    * This seems preferable as Gymnasticon no longer needs to know about wheel circumference.
    * Wheel revolutions data can be passed through unmodified.
  * Feature should be off by default and enabled via CLI/config. This is to avoid confusing the average user. Typically the speed is calculated by Zwift (or another app) and the value there will be very different from the speed value coming from the bike.
  * It could be useful for the "bot" to also support speed for testing purposes.
    * Seems to require an approach similar to option 1 above.
    * Only if it's easy to add and doesn't complicate things.

I plan to look at this but I have limited time right now. I'll of course accept a PR if anyone wants to jump on it sooner, and will offer any help/feedback needed!

nealjane, 2021-03-31:
@djfjeff have you tried these? State they provide ‘power’ to non power Garmin watches...

https://apps.garmin.com/de-DE/apps/c7a333a6-278b-47c6-b65b-205ec484150c

Or 

https://apps.garmin.com/en-US/apps/70f1aea3-f1d3-47f6-b7aa-23f029a75e3b#0

djfjeff, 2021-03-31:
@nealjane yes, they might work but the power field is outside the Garmin world, in a sense, it is not used by anything Garmin (training status, training load, etc). It is useful to show the power on non-power watch however.

bthaler1, 2021-04-15:
I'd just like to express my interest as well for adding speed data to the ANT+ and BLE outputs. I also use Gymnasticon without Zwift, connected to a Peloton. I'm not sure about the IC4, but for Peloton it looks like speed can be calculated as a function of power like this:
Let r = sqrt(P).
For P < 26, S = 0.057 - 0.172 r + 0.759 r^2 - 0.079 r^3
For P >= 26, S = -1.635 + 2.325 r - 0.064 r^2 + 0.001 r^3
(Copied from https://ihaque.org/posts/2020/12/25/pelomon-part-ib-computing-speed)

screetch82, 2021-05-11:
Hi
thx for the great tool. I used this on an IC4 and also a Garmin watch to record all my stats. 
Now Garmin can connect to IC4 via BLE to get Speed, Distance, RPM i think. However Watts is only recorded in Garmin when
it  is provided via ANT+ 

So when I used gymnasticon I connected the IC4 to Gymnasticon instead of the garmin watch. and added the cadence and power as separate sensors in garmin. However speed is indeed now missing as a separate sensor. 

It would be good to just output the value the IC4 actually transmits via Ant+  
This can be added as a third separate sensor in Garmin for the people who need it. 

See screenshot that the Power output finally enables some cycling statistics in Garmin:

[image]

screetch82, 2022-01-01:
HI, can confirm it works fine with Garmin . I used my Garmin edge to add power and speed/cadence sensor from Gymnasticon

[image]