ptx2/gymnasticon #77: Stop broadcasting when the bike is turned off

djfjeff, 2021-03-22:
The issue I have is that Gymnasticon always broadcast the sensors, even when the indoor bike (IC4) is OFF and disconnected.

The issue it cause is when I get on my road bike outside and start an activity, my watch pick up and link with my indoor bike Power sensor. So, my Garmin watch start to use this Power sensor for the activity instead of using my actual road bike power sensor.

Also, everytime I start an activity, I get the message that the Power sensor is connected, even for a walk or run activity.

Having Gymnasticon stop broadcasting when the bike disconnect would solve this issue.

nealjane, 2021-03-22:
Maybe it’s specific to that bike - As far as I know - my pi gymnasticon stops broadcasting when my keisers off.

alternative fix = turn off you pi when not using it?

djfjeff, 2021-03-22:
Turning off my Rpi will not be ideal as I run the read-only version on a Zero W. So it would imply unplugging and replugging the Rpi everytime. Not impossible but far from ideal.

It might effectively be a bug with the IC4 bike if it works with your Keiser bike.

djfjeff, 2021-03-22:
I did some more testing and it seems that only the Power ANT+ is still broadcasting after the bike turn off. The BLE Cadence is effectively stopped.

ptx2, 2021-03-23:
Thanks for reporting this @djfjeff!

commit 89578658 (2021-03-28): Stop ANT+ server on exit (#80)

Fixes #77.
closed 2021-03-28 by ptx2