ptx2/gymnasticon #78: Installer fails on Mac OS (Big Sur) with JS error

francisjervis, 2021-03-24:
Hi, I'm getting this when I run `gymnasticon` after installing:

```
/usr/local/lib/node_modules/gymnasticon/node_modules/@abandonware/noble/lib/resolve-bindings.js:11
    return new (require('./mac/bindings'))(options);
           ^

TypeError: require(...) is not a constructor
    at module.exports (/usr/local/lib/node_modules/gymnasticon/node_modules/@abandonware/noble/lib/resolve-bindings.js:11:12)
    at Object.<anonymous> (/usr/local/lib/node_modules/gymnasticon/node_modules/@abandonware/noble/index.js:2:51)
    at Module._compile (node:internal/modules/cjs/loader:1092:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1121:10)
    at Module.load (node:internal/modules/cjs/loader:972:32)
    at Function.Module._load (node:internal/modules/cjs/loader:813:14)
    at Module.require (node:internal/modules/cjs/loader:996:19)
    at require (node:internal/modules/cjs/helpers:92:18)
    at Object.<anonymous> (/usr/local/lib/node_modules/gymnasticon/lib/app/app.js:16:37)
    at Module._compile (node:internal/modules/cjs/loader:1092:14)

```

ptx2, 2021-03-24:
Hi, which node and npm version do you have there?

```
node --version
npm --version
```

Try upgrading to latest npm (7.7.1) and do `npm install` again with a fresh clone.

The error itself looks like a bug introduced in version 1.9.2-12 of `@abandonware/noble`. However npm should install version 1.9.2-10 of noble, per package-lock.json. Older versions of npm won't use the package-lock.json. I've confirmed that npm 6.13.4 and 7.7.1 both install the correct version of noble on a fresh clone on Big Sur.

Good luck! Let us know what happens and we can update the README with this info.

francisjervis, 2021-03-24:
I upgraded to the latest npm (7.7.3) and unfortunately still get the same error. That is after running `npm uninstall gymnastico`n and reinstalling.

ptx2, 2021-03-24:
Ah, I see the problem now. Try this:

```
npm uninstall -g gymnasticon
npm install -g gymnasticon@1.4.1
```

francisjervis, 2021-03-25:
This worked to clear the install bug, thanks. Opened another issue however...

closed 2021-03-25 by ptx2