ptx2/gymnasticon #79: Does not connect to Flywheel from Mac.

francisjervis, 2021-03-25:
Hangs at "connecting to bike" - Derailleur connects to it (I used it to calibrate), the FlyTest page also does not see it, which I assume runs off the same underlying code. Can post debug logs from Derailleur if that's useful.

ptx2, 2021-03-25:
It's possible there are multiple issues at play, but it sounds like the bike might just not be advertising because it's off/asleep or something else is still connected to it.

Some easy things to try:

1) Make sure any other apps that can connect to the bike like Derailleur and FlyTest are closed.
2) Check the macOS Bluetooth menu and disconnect "Flywheel 1" if you see it connected. The Bluetooth menu can be enabled thru Bluetooth in System Preferences.
3) Check any other devices (phone/tablet) that might be connected to the bike.
4) Turn the crank while it's trying to connect to keep the bike awake and advertising.
5) Batteries and Bluetooth range are the most common issues, but since you can connect with Derailleur we can rule those out.

One thing that could prevent Gymnasticon from finding the bike is if its advertised BLE name does not start with "Flywheel". If you find that your bike has a different name, you can use `--flywheel-name` to disable the name check (pass it `""`) or target an exact name. The reason it filters by name is because many other devices use the same service uuid (Nordic UART Service). As far as I've seen, every Flywheel bike has the name "Flywheel 1" so I'd be surprised if this was the issue, let us know if it is!

By contrast, FlyTest doesn't filter by name and instead lists all Nordic UART Service advertisements (including the Flywheel bike) in the browser's BLE dropdown. (They use different BLE implementations under the hood)

ptx2, 2021-04-21:
Hey @francisjervis I'm going to close this out as it's been a while. Please feel free to reopen if you're still having issues!

closed 2021-04-21 by ptx2