ptx2/gymnasticon #83: Supporting a Keiser M3

ajs123, 2021-04-03:
I have a Keiser M3 which is very close to the M3i only without Bluetooth. I am considering adding support for it, which would involve

- bringing the flywheel speed sensor (switch closure) to GPIO pins and timing the intervals between rotations
- doing something with the resistance sensor, which is a potentiometer attached to the resistance magnet assembly. This could be an added A/D converter or more likely add a circuit so the pot controls the frequency of another pulse train like the flywheel

Would there be any fundamental flaws (or serious challenges) to this approach? The pedal to flywheel ratio is 8.75. Getting cadence from flywheel pulses would require measuring the pulse interval with about 1ms resolution. Or pulses might be counter over a brief period. 

Thanks for any comments!

nealjane, 2021-04-03:
@ajs123  there are loads of m3 users looking for ways to upgrade their bikes for Bluetooth. I bought one of these when they were available on Amazon.co.Uk (up till Xmas 2021) now only available in spain which is a shame as it’s working great on my old m3.
https://www.amazon.co.uk/gp/aw/d/B01MPZLUG1/ref=ppx_yo_mob_b_inactive_ship_o0_img?ie=UTF8&psc=1

nealjane, 2021-04-03:
https://m.youtube.com/watch?v=Slyz55tzwvQ

ajs123, 2021-04-03:
Yes - a shame for sure! I’d seen this and hit the same “not available” dead
end.  Of course, Gymnasticon wired in would add capabilities and avoid
having to get Keiser’s M3 adapter. I’m guessing that the DPCycling unit is
Arduino based.

ptx2, 2021-04-04:
Hi @ajs123,

That sounds very doable!

The [pigpio](https://github.com/joan2937/pigpio) library should work for the counting (it can give timestamped pin change events with sub 5-microsecond accuracy). Here's an example using the pigpio JavaScript bindings: https://github.com/fivdi/pigpio#determine-the-width-of-a-pulse-with-alerts

There are off-the-shelf ways to add an ADC to the Pi if you wanted to go that way. Search for ADC "HATs" on Digikey, Mouser, Pimoroni, DFRobot, Tindie, Adafruit. Here's a JavaScript library for using MCP SPI ADCs on the Pi: https://github.com/fivdi/mcp-spi-adc

BTW have you already poked around the display/computer PCB? I'm wondering if you could get the power/cadence/resistance data out from there somehow e.g. a hidden uart debug interface, or if the display comms could be sniffed?

nealjane, 2021-04-04:
Came accross this today - may give you some more info m3 setup.

https://www.bikeforums.net/training-nutrition/519130-kaiser-m3-true-wattage.html

Here's a quote from a Keiser Rep on how they calculate power taken from another forum (https://www.innercycling.com/f/showth...=8542&page=2):

"John,
First and foremost the M3 is not a ergometer nor is it priced like one. Power is an estimate. The price and our maximum usage (group exercise) dictate that it is not for testing and it cannot be calibrated. Though we do have some people that have used it with an IMET test when a bike is more preferential than a treadmill. One thing not mentioned below is magnet strength, they are consistent and do not weaken over the life of the bike. 
To answer the question:

Watts are calculated from the gear setting. A potentiometer is attached to the magnet holder (the round cone shaped disc at the end of the shifter cable). As the shifter is moved, the cable rotates the magnet holder. A potentiometer is rotated by the rotation of the magnet holder, thus feeding information to the computer on the position of the magnet holder. The rotation of the potentiometer is broken down into 24 gear settings. A dynamometer was used to test and develop a table of wattage at various gear settings and speeds. The speed of the crank is determined by a magnet attached to the large pulley on the right crank arm and a magnetic switch attached to the circuit board in the magnet holder assembly. Each time the magnet on the pulley passes by the magnetic switch, a signal is sent to the computer to compute the RPM's of the crank arm. Power equals force times velocity. The force is determined by the magnet position and the speed by the crank speed. The lookup table is programmed into the computer and the computer simply looks at the gear setting and speed and goes to the lookup table to find and display the Watts for those two settings.
Darrin Pelkey
VP Sales & Marketing at Keiser
The Power in Human Performance"

ajs123, 2021-04-04:
Yes - very helpful! The choice of using a lookup table is interesting: It
could be that the math is an issue for the CPU, or simply not bothering
with modeling torque as a function of resistance setting and speed. I did
embedded controller design decades ago with 8-bit CPUs running in the 1 MHz
range and we thought about such things all the time. The take-home point,
though, is that torque depends upon both resistance setting and speed,
which is both unsurprising and good to know!

nealjane, 2021-04-04:
[image]

nealjane, 2021-04-06:
@ajs123  Found this today!!
https://github.com/turbodonkey/bike_power_meter

ajs123, 2021-04-06:
This is extremely helpful. Thank you.

From other reading, torque from an eddy current brake can be modeled as the
product of a magnetic flux-dependent (magnet position) function and a
speed-dependent function. The Keiser illustration provides the
position-dependent function. The speed dependence should be easy to
determine using the Keiser display as a data source.

It's curious that the "gears" aren't evenly spaced!

ajs123, 2021-04-06:
Thanks! My last decision before proceeding is

RPi building on your work:
+ Nice live console development environment
+ BLE and WiFi at the same time - potential to implement a web-based
enhanced display
-  Needs A/D hat or an add-on oscillator
-  Limited or no operation without wall power

Microcontroller
+ Should have decent operation on battery
+ As little as zero added hardware (depends on how the Keiser computer
terminates the position pot)
-  No modules with simultaneous BLE and WiFi (some kludgy-looking schemes
to switch back and forth)
+/- Return to past life of microcontroller development cycles

This project - https://teaandtechtime.com/arduino-ble-cycling-power-service/
and https://github.com/Tschucker/ArduinoBLE-Cycle-Power-Service is another
example.

ptx2, 2021-04-21:
Hey @ajs123 sorry for the slow reply here.

I went through a similar decision when starting this project. The reason I chose the Pi is because it's versatile (lots of I/O options), widely available, and relatively straight forward to setup for a non-tech user.

My gut says microcontroller is probably easier, particularly because of the battery requirement. I'd love to see this pulled off with a Pi though :-)

Interested to know which way you end up going. Sounds like a very fun project.

ajs123, 2021-04-25:
I've gone with a microcontroller - using an Adafruit nrf52850 Express,
based on a Nordic controller with Bluetooth (though no WiFi). I *think* it
can also do ANT+ and Bluetooth at the same time but haven't looked
carefully at that.  As of today, I have it connected to the Keiser bike,
reproducing Keiser's "gear" and power readings, and providing data to
cycling apps using either Fitness Machine Service or Cycling Power Service.

I've tested using Kinomap and all seems to be good. For Cycling Power
Service, I'm reporting resistance in addition to pedal strokes and power,
and Kinomap doesn't seem to use it, but I think that's normal for Kinomap.
If connecting with Fitness Machine Service, I think Kinomap wants to change
the resistance even though I say that feature's not there, but again that
may be a Kinomap thing.

I'm currently cleaning up the code and need to take some time to transition
from prototype in an Airpods box to something "real."

I'm happy to share my calibration data, info on the Keiser hardware, etc.!

nealjane, 2021-04-26:
@ajs123 Thats fantastic news - you need to be sharing this with your fellow  American M3 users!

skoregon, 2021-04-26:
@ajs123 Would love to see your Adafruit board code if you'd set up another repository.

ajs123, 2021-04-27:
Yes - as soon as I clean it up a little!

robcalm, 2021-04-27:
This is cool.

Somewhat related... I've been dreaming of having an actuator to control a second magnet set in response to terrain changes.  I think an associated potentiometer wired in series with the pot on the existing magnet system could be made to work somehow.  Maybe this is something for the future.

Maybe the smartspin project unit could be modified to do the adjusting, or maybe there is some more clever way in line with what you're doing here.
https://github.com/doudar/SmartSpin2k

ajs123, 2021-04-27:
Definitely related. When I realized that the Fitness Machine Service expects to set the resistance, I thought, OK, I need a servo on the resistance.

On the M3, a servo on the existing resistance lever (loosening the center screw reduces the force to cable friction plus the spring in the magnet assembly) might be easiest and avoids having to run another wire. From what I’ve seen so far, if there were two magnet assemblies it would be hard to avoid needing to know where each of them is.

I will clean up the code and post.  I can also share my calibration data though it was taken a bit hurriedly!

ajs123, 2021-04-29:
OK, folks - here it is.  https://github.com/ajs123/KBikeBLE

ajs123, 2021-05-19:
@nealjane, where did you find [the Keiser graph on power estimates](https://user-images.githubusercontent.com/68538658/113517760-e75ac880-9579-11eb-968e-854193421594.jpeg) that you posted? I've made use of it and want to properly cite it if I can.

I think I've figured out how to calibrate the M3 using Keiser's calibration tool. I'll post details over at [KBikeBLE](github.com/ajs123/KBikeBLE) when I have a chance. I can describe it here as well if there's interest.

Thanks!

nealjane, 2021-05-19:
@ajs123 it was just posted on the keiser m3i Facebook group - so no idea where it originates from.

Keiser tool is £25 here in Uk - so I modelled my own version recently (doing the tooless version appears to not work and will mess up the gear vs effort reqd)

theres 2 versions - calibrating older than 2009 differs and needs a different tool! See here-
https://youtu.be/R1CSYF7tZeg
So I modelled a best guess version for that one too (magnet set to edge of wheel rather than 10mm in on the more recent tool)

ajs123, 2021-05-19:
Thanks, @nealjane. Thanks @nealjane. Here's what I've found on a 2009 M3:
- Used https://apps.automeris.io/wpd/ to digitize the chart, taking (resistance, power) values at the "bottom" of gear 1, the boundaries between gears, and the "top" of gear 24. 
- Switched back and forth between a Keiser computer that had been calibrated to the bike and my computer, to relate my ADC readings from the pot to the x-axis on the graph.  
- At each spot, pedaled at 90 rpm and got a power reading from the Keiser computer. This simply verifies that the chart accurately shows the Keiser power estimate.
- I also got ADC readings for the the magnet assembly at its low-end stop (cable unhooked or adjusted to allow the assembly to hit the stop), with the cal tool, and at brake pad contact. The cal tool is obviously key in Keiser's procedure for their 4.06 software. The other positions are probably not (see below).

Findings:
- If the x-axis on the chart is re-labeled with 0 as the lefthand edge of gear 1 and 100 as the righthand edge of gear 24, the cal tool position is 40%. I'm in the process of updating KBikeBLE to use this scale.
- The cal tool is also 40% if the mechanical stop on the magnet holder is -5 and brake pad contact is 110. This could be bike-dependent, and also dependent upon whether the brake pads wear at all. I neither imagine that brake pads get used much, nor expect that Keiser depends upon them being precisely uniform.

Further thoughts
- I'd started out thinking that calibration requires two points to define the line relating ADC values to magnet position. Keiser can't use the low end position because their calibration procedure doesn't require that the shift lever allow the magnet assembly to travel all the way to the bottom. They could use brake pad contact, but that wouldn't seem to be super reliable.
- While the magnet position potentiometer is +/- 10%, as long as it's being used as a voltage divider, any other resistances in the circuit are low enough, and the rotational range of the pot is consistent from unit to unit, the change in reading per degree of magnet movement should be consistent as well. This seems like a lot of "ifs" but I can see them all being true. If so, the slope of the line relating ADC readings to the x-axis (I'm calling it Resistance in %) will be consistent from unit to unit.
- That leaves the intercept or offset. This is clearly dependent upon the potentiometer and gear position when the magnet cover is assembled.
- Putting all of this together, Keiser's software would
   - After one crank revolution after waking up, look for the largest magnet position. This should be contact with the cal tool. It provides a measurement of the offset.
   - After one more crank revolution, look for the lowest and highest values. They should merely be consistent with somewhere around the bottom of gear 1 and brake pad contact. It's a "sanity check" on the cal tool reading.
   - Registering a new calibration only under very specific conditions including the 6 full swings after the second crank rotation and within 30 seconds guards against any user doing it unintentionally.
- The YouTube videos showing calibration without a tool have you set the gearshift lever to the vertical position. This is pretty _close_ to where it would be at the cal tool position, so the unit passes the "sanity check" but is calibrated to the lever position rather than the magnet position.

Regarding replication of Keiser's power estimates, I've verified (reasonably well) that it works to multiply Keiser's curve by a factor that depends upon the cadence. That factor is pretty close to linear with RPM from 60 to 110, though I'm using a quadratic which does a tiny bit better in that range and comes nicely down to zero at 0.  Keiser's curve is nicely captured by a quadratic only from gears 2 through 20 or 21. Capturing the whole thing with an equation requires a 4th order polynomial. Though the nrf52840 I'm using has hardware floating point, I'm planning to switch to a lookup table with interpolation. The same table can be used to replicate Keiser's unevenly spaced gears for those who want that. My wife rides using the gears because she's accustomed to them. I use the %resistance display.