ptx2/gymnasticon #85: Keiser output/speed over ANT+ to Garmin Venu

hascherpur, 2021-08-11:
Thank you for your great work. I just got gymnasticon running and it is working well. I additionally bought an ANT+ dongle that I include when running gymnasticon. However, I notice that my Garmin Venu cannot pick up the output/speed over the ANT+ connection. When starting gymnasticon, it does say that is opening the ANT+ device, but again I cannot seem to get the wattage and speed. Any ideas?

hascherpur, 2021-08-13:
Thanks for the quick reply. Damn, it is a shame that my watch does not support a power profile. I tried the ant+ power meter, but could not get it to find my ant+ output. Not sure how I could detect my ant+ id for the app.

nealjane, 2021-08-13:
You have to put in the ant if for it to work - it’s  11234

nealjane, 2021-08-13:
Need to add the ant Id in the power app settings for it to work - 11234

hascherpur, 2021-08-13:
Thanks for the solution and help! It worked!

closed 2021-08-13 by hascherpur