ptx2/gymnasticon #86: Available metrics for garmin Edge

byndcivilization, 2021-08-23:
I have a Garmin Edge 320 that I'm using to record things. With a little trial and error, I've gotten gymasticon to register the pi as an ant+ power sensor and during rides,  the Edge reports back power and cadence. Is there a trick to getting speed out as well or does that not work here.

byndcivilization, 2021-08-23:
I should mention that im using a peloton bike. I don't see speed metrics being picked up there. So im guessing its not implemented for the peloton.

nealjane, 2021-08-23:
Hi, yes- speed isn't implemented for any bike. 
If you had to choose between speed vs power - you're better off with power, 

apps like zwift will calculate your speed and distance if you want those stats - sync them to Strava and then sync that to Garmin connect perhaps??

byndcivilization, 2021-08-23:
Rad, I mostly I wanted to make sure I had everything working as expected.

closed 2021-08-23 by byndcivilization

adanemayer, 2022-03-18:
@byndcivilization Can you share the details of how you got the Edge to pair with the ANT+ USB dongle? I've tried now with both a cheap one and one from Garmin. I have a Garmin Edge 25 I am trying to pair and it doesn't see any available devices with either dongle plugged into the Raspberry Pi that is linked via Bluetooth to my Schwinn IC4. 
I confirmed the Raspberry Pi + IC4 work by pairing my laptop to the Bluetooth signal from Gymnasticon and watching the data stream through so I think it is just a matter of getting the ANT+ registered.

nealjane, 2022-03-18:
@adanemayer . You Garmin edge 25 doesn’t do power profile. It will do cadence and speed so you need to use Chris elsens update to Gymnasticon - https://github.com/ptx2/gymnasticon/pull/89

this update adds speed and cadence profile over ant+ which your Garmin will be able to connect to.