ptx2/gymnasticon #87: No power metrics, only cadence

Lvrebel, 2021-09-13:
Just finished set up and only cadence is coming through. Zwift allows for connection to Gymasticon for both power and cadence but no power metrics. Also confirmed same issue on another app. Is it possible that low D batteries are causing this issue?

cagnulein, 2021-09-13:
Did you calibrate your bike?

Roberto Viola
Software engineer and open source enthusiast
http://robertoviola.cloud

Lvrebel, 2021-09-13:
I just read a closed issue mentioning the same thing. I will work through the calibration. When I try to calibrate using Flywheel Test webpage, nothing happens even with iOS installed. I may try a Mac or PC laptop.

becker53, 2021-12-22:
I have a similar issue. My bike is calibrated. I can read both power and cadence on the flytest app and on the icg app independent of this. Using gymnasticon I can read both power and cadence on any virtual cycling app, but I can only get cadence on my Garmin. When adding new devices my Garmin will only find the cadence sensor.

nealjane, 2021-12-23:
@becker53 are you certain your Garmin does Bike Power profile - which Garmin is it??

becker53, 2021-12-23:
Yes. It is the forerunner 945. I can connect it to cycling power meters and also to power meters on smart spin bikes at the gym.

becker53, 2021-12-23:
One more thing, I normally connect via ant+ but I did just verify that my typical power connections work via ble as well.

nealjane, 2021-12-23:
Ant+ works for power and cadence, 
the BLE connection to your watch WILL only show cadence.

becker53, 2021-12-23:
Do you know what the cause of that is? I just verified that I get power via ble from my pedals.

nealjane, 2021-12-23:
If you’re only seeing the cad/Spd xx.xx.xx.xx.xx then that’s the Bluetooth. (Which you should not use with your Garmin as it only shows cadence - I dont know why - it’s just how it’s setup I guess). 
ant+ was implemented specifically for connecting to Garmin watches and bike computers - and so will work - perhaps your specific ant+ dongle is the issue ?? Get as close as you can to it with your watch in case it’s just interference as it’s more prone to it vs Bluetooth..??

becker53, 2021-12-23:
I do not have an ant+ stick. Since I can get power from my pedals via ble I figured it would be fine. I guess I will have to get one. Is it as simple as plugging the stick into the pi? Do I need to setup anything else, like in the config file?

nealjane, 2021-12-23:
Yes, just plug it in and restart pi and it will work, you’ll need to usb to micro USB adapter too.

becker53, 2021-12-23:
Thanks so much. I will give that a shot. I just remembered that according to https://ptx2.net/posts/unbricking-a-bike-with-a-raspberry-pi/ ant+ was not supported and so didn't even bother. I really appreciate you help a lot.

nealjane, 2021-12-23:
Will definitely work with ant+, I’d suggest staying away from the cheaper anself type dongles with black ribbed case (some of them aren’t fully compatible with ant protocols supposedly- although they will work I found I had drops in connection during rides, I use Garmin one (or livlov/chileaf -same item with black casing with red strip on work fine too). 👍🏻

becker53, 2021-12-23:
Just to be sure, you do like the livlov with the red stripe? Just checking because it also has the ribbed case. I have heard good things about anself but it also has the black ribbed case. The ones I am looking at are mostly the same but it is hard to tell any differences.

nealjane, 2021-12-23:
Anself ones are supposed to be fine, the issue is the dodgy types look exactly the same (black ribbed) and I’ve read online some knock off chinese ant+ were even rebranded to Anself - so, I guess you can take the chance it’s a legit Anself- if it doesn’t work you buy it from the likes of Amazon who take returns (and not eBay).

livlov/chileaf with red stripe ive used and have worked great.

nealjane, 2021-12-23:
> mostly the same but it is hard to tell any differences.

That’s why I suggest you stay away from the black ribbed ones.🤣

becker53, 2021-12-23:
Ok. Good to know. Thank you for educating me.