ptx2/gymnasticon #93: Ant+ dropouts with Keiser new ‘M connect’ computer introduced mid ‘21 onwards

nealjane, 2022-02-18:
newly introduced Keiser computer mid 2021 (upgrade available to users from mid 2021) standard on 2022 us bikes I believe.

dropouts in Power over ant+ seen - (previous ant+ position fine prior to computer upgrade - and even positioned next to watch still saw drops in signal) 

Typical power over ant+ Gymnasticon connected to the newer mid 2021/2022 Keiser ‘M connect’ computer from last autumn (prior to Gymnasticon speed implementation) connected to fenix 6
[image]

this was using power and cadence from Gymnasticon and speed cadence from the new Keiser computer over bluetooth.

As Gymnasticon now provides speed - it’s whether it was the speed connection over Bluetooth from new Keiser computer that causes interference?? 
(I don’t think I tested it without Keiser speed connected). Def Looks like I need to swap over to my newer computers again to see if problem still persists for newer Keiser M3i computer users. 
I’m off work as it’s half term so will try this week.

nealjane, 2022-02-27:
So - please note this only relates to New Keiser M connect computer intro 2021/22.

Connecting Gymnasticon ant+ to fenix for p/c/s.
Connecting Gymnasticon to power in zwift.

intermittend issue where all stats drop to 0 at same time (ant and bl) - only way to reset is to go to menu on zwift which seems to reset connection.

M Garmin connect stat shows where it drops out over P/C and speed same. 
[image]

Zwift analyser indicated no disconnection within same time frame 
[image]

- so makes me think it’s something to do with the new mix of Keiser computer and gymnasticon. 

only suggestion is to use the new Keiser computer for connecting to zwift, and use gymnasticon ant+ only which seems to solve the issue.

nealjane, 2022-03-01:
Quick test again today Ant+ donegle repositioned on 2m extension cable (so 2m away from Raspberry pi) and 1m away from new Keiser M connect computer to see if it helped (I think its the new computer TBH as this never happens on older computer for me) - Note it took 20 seconds to refund the signal at [2022-03-01T15:16:52.819Z] which is the dropouts being seen in ANT+ on garmin watch as well as on swift (even though Zwift says there's no dropout in signal when checked - just drops to 0)

> [2022-03-01T15:16:50.841Z] received stats from bike [power=30W cadence=70rpm speed=15.54km/h]
[2022-03-01T15:16:50.875Z] wheel rotation [timestamp=1646147810875.354 revolutions=72 speed=15.54km/h power=30W]
[2022-03-01T15:16:51.047Z] pedal stroke [timestamp=1646147811047.4243 revolutions=42 cadence=70rpm power=30W]
[2022-03-01T15:16:51.362Z] wheel rotation [timestamp=1646147811360.9138 revolutions=73 speed=15.54km/h power=30W]
[2022-03-01T15:16:51.848Z] wheel rotation [timestamp=1646147811846.4736 revolutions=74 speed=15.54km/h power=30W]
[2022-03-01T15:16:51.905Z] pedal stroke [timestamp=1646147811904.5671 revolutions=43 cadence=70rpm power=30W]
[2022-03-01T15:16:52.333Z] wheel rotation [timestamp=1646147812332.0334 revolutions=75 speed=15.54km/h power=30W]
[2022-03-01T15:16:52.762Z] pedal stroke [timestamp=1646147812761.71 revolutions=44 cadence=70rpm power=30W]
[2022-03-01T15:16:52.819Z] wheel rotation [timestamp=1646147812817.5933 revolutions=76 speed=15.54km/h power=30W]
Stats timeout: Restarting BLE Scan
[2022-03-01T15:16:52.875Z] received stats from bike [power=0W cadence=0rpm speed=undefinedkm/h]
[2022-03-01T15:17:08.690Z] received stats from bike [power=3W cadence=25rpm speed=3.62km/h]
[2022-03-01T15:17:08.710Z] pedal stroke [timestamp=1646147828692 revolutions=45 cadence=25rpm power=3W]
[2022-03-01T15:17:08.715Z] wheel rotation [timestamp=1646147828693 revolutions=77 speed=3.62km/h power=3W]
Stats timeout: Restarting BLE Scan
``

nealjane, 2022-03-02:
https://github.com/ptx2/gymnasticon/pull/89#issuecomment-947848551 - totally forgot that we'd had a discussion about this previously @chriselsen . 

Have had a mess about/tweaking and for the life of me haven't managed to find any way of imroving this issue.
So new keiser m3i computer still has old BLE shown as 'M3', but now also has M3i#000 (3 dig numbers based on bike number allocated) - this new bluetooth has Cycling speed and cadence and fitness machine.  NRF Connect screenshots attached.
[image]
[image]
[image]
[image]
[image]

Bike will connect ok to computer - but every no wand then will loose connection between bike and Gymnasticon and can take up to 60 seconds (if it does eventually reconnect)..
[image]

`Keiser M3 bike version:  6.46  (Stats timeout:  2  sec.)
[2022-03-02T15:49:58.401Z] bike connected d0:1e:2f:c4:ff:bc
[2022-03-02T15:49:58.496Z] ANT+ stick opened
[2022-03-02T15:49:58.862Z] received stats from bike [power=4W cadence=31rpm speed=4.41km/h]
[2022-03-02T15:49:58.885Z] pedal stroke [timestamp=1646236198868 revolutions=1 cadence=31rpm power=4W]
[2022-03-02T15:49:58.894Z] wheel rotation [timestamp=1646236198872 revolutions=1 speed=4.41km/h power=4W]
[2022-03-02T15:49:59.368Z] received stats from bike [power=4W cadence=26rpm speed=4.41km/h]
[2022-03-02T15:49:59.879Z] received stats from bike [power=4W cadence=26rpm speed=4.41km/h]
[2022-03-02T15:50:00.584Z] wheel rotation [timestamp=1646236200583.0205 revolutions=2 speed=4.41km/h power=4W]
[2022-03-02T15:50:00.898Z] received stats from bike [power=3W cadence=26rpm speed=3.62km/h]
[2022-03-02T15:50:01.177Z] pedal stroke [timestamp=1646236201175.6924 revolutions=2 cadence=26rpm power=3W]
[2022-03-02T15:50:01.408Z] received stats from bike [power=3W cadence=26rpm speed=3.62km/h]
Restarting BLE Scan
[2022-03-02T15:50:02.427Z] received stats from bike [power=2W cadence=20rpm speed=2.78km/h]
[2022-03-02T15:50:02.937Z] received stats from bike [power=2W cadence=20rpm speed=2.78km/h]
[2022-03-02T15:50:03.298Z] wheel rotation [timestamp=1646236203297.2651 revolutions=3 speed=2.78km/h power=2W]
`

nealjane, 2022-03-02:
Its most definitely auto connecting to ic4 when it drops - so, quick and dirty solution is to either remove ic4 from the autodetect, or just remove the auto detect entirely I guess! ( adding in the gymnastics.json doesn't appear to override the autodetect)

nealjane, 2022-03-04:
Removing ic4 profile etc hasn’t helped at all, reading on fb groups others having dropouts using Keiser BLE connection (old m3 BLE) - so it’s looking more like an issue with the new computer rather than gymnasticon. (Although no dropouts seen using the keiser app to connect)

nealjane, 2022-03-18:
Right, I’ve been testing this for weeks now (on pi zero 1) it’s doesn’t appear to be gymnasticon but either the pi’s /the new Keiser hardware that has the issue. Tested it with zwift connected to new Keiser computer (M connect) which sends out fitness bike profile over BLE, Gymnasticon connected to Keiser direct BLE connection (standard connection) and sending the ant+ to my Garmin watch. 

On pi zero (v1) and it’s working pretty much flawlessly. Changes I made, full apt update gymnasticon Keiser file I set dropout from 2s to 3 and set wifi (mainly to alter Bluetooth Channel’s) to JP in config which seems to have sorted it. Only minor drops seen in image-
[image]

Using the same sd card in a pi zero 2 - it’s still dropping out for 30s at a regular intervals - 
[image]
Tried apt update all and nothing has improved over ant+/Bluetooth.
@chriselsen - any other suggestions to try?? It’s clearly related to pi hardware..

chriselsen, 2022-03-18:
The only idea I could come up with is to use a USB extension cable and move the ANT+ stick away from the Pi and its Bluetooth/Wifi antennas.

I usually do that anyways to brink the ANT+ stick closer to the bike handle, where my wrist with the Garmin watch is and have pretty much no connectivity issues with ANT+ that way.

nealjane, 2022-03-18:
Funnily enough that was the first thing I tried when I saw the issue as I assumed it was interference from the additional Bluetooth. Made no difference unfortunately with a 2m long extension for ant+. Also moved pi away from bike computer etc. 

connecting pi and the keiser app to the bike (both using same Keiser direct BLE - and when pi drops to 0 for 30s the app is still working showing correct stats so Keiser computer is definitely still sending data - so it looks like it’s a pi02 issue..(checked this when I saw this on the Keiser website) - 
[image]

nealjane, 2022-03-19:
Ok I’ve got it working on pi02, - 7s seconds works with new M connect computer.

Keiser M3 bike version:  6.46  (Stats timeout:  7  sec.) any less and the 30s dropouts start..
[image]

const KEISER_STATS_TIMEOUT_NEW = 7.0

closed 2022-03-19 by nealjane