ptx2/gymnasticon #94: Can't connect Zwift to Gymnasticon

maustin76, 2022-03-06:
I'm sure I'm missing something obvious, but I've been beating my head against the wall for the past 2 days and have come here begging for help :)

1) bike: Flywheel IC5
2)  Image  (v1.4) from Github loaded onto SD card (using Raspberry Pi Imager 1.7.1) and installed in Raspberry Pi 4b
3) Gymasticon appears to be working (I can see "Gymnasticon" as a bluetooth option on my phone)
4) I can successfully calibrate the bike (which shows as "Flywheel 1") via the web app (using an android device)
5) In Zwift (using an android device) I CANNOT get gymnasticon to show up as a sensor (specifically, I'm trying to add it as a power sensor)

troubleshooting attempted
a) Scoured the internet, found almost nothing relating to my issue (which is usually a pretty good indicator that I'm doing something wrong)
b) I've tried two independent SD cards, two RP 4bs, and two andriod devices (running zwift) to eliminate possible hardware issues
c) Even though I am able to calibrate the bike, I still swapped out the D batteries just to be safe
d) Of note (maybe?): I can not calibrate the bike while Gymnasticon is running (I assume that implies the bike is busy talking with Gymnasticon?); as soon as I turn Gymnasticon off (i.e., remove power to the pi), "Flywheel 1" pops up in the calibration app (and rpm, power, and resistance all vary correctly on the app display after calibrating).
e) I've tried physically relocating the bike, Pi, and android devices (maybe the Pi is too far from the bike?  Maybe it's too close to the bike?  Maybe the moon is in the wrong phase??)
f) I connected a keyboard/mouse to the Pi; logged in using pi/raspberry ... then realized I don't speak Linux... nor Pi... so I ventured no further. 
g) Tried installing v1.3 (grasping at straws)

Disclaimers:

1) I have almost no Pi experience 
2) I have never used Zwift before this trial

ANY help would be greatly appreciated -- thanks in advance!

v/r,

Mike

nealjane, 2022-03-06:
Hi Maustin- have you tried an alternative app (in case it’s a zwift issue)?? RGT or Sufferfest etc… 

did you allow all privacy settings on installing zwift (eg Bluetooth etc?) try uninstalling and reinstall app..

nealjane, 2022-03-06:
You haven’t Paired gymnasticon in your phone/tablets system settings have you?? - because you only need to pair in the app eg zwift..

maustin76, 2022-03-07:
Thanks @nealjane!  Great ideas.  I haven't solved my issue yet, but I'm closer (and suspect it's a Zwift issue).

I had originally planned to installed the Wahoo app (Systm)  to further troubleshoot, but it required subscribing (i.e., putting in my CC info) before starting the trial, so I didn't follow through with it.  I assumed that installing Zwift on two separate devices (including one re-install) would be sufficient to eliminate Zwift, but clearly I was wrong :)  

I installed Systm today and it immediately recognized Gymnasticon.  I still can't get my android devices to work with Zwift, but now that I know Gymnasticon seems to be working properly, I'll focus my efforts on Zwift.  

Again, thanks for the help!  (BTW if you have any tips on getting Zwift to work with Android devices, I'm all ears :)  -- Samsung S20 Ultra and/or Samsung Galaxy Tab S6).

v/r,

Mike

nealjane, 2022-03-07:
> Thanks @nealjane! Great ideas. I haven't solved my issue yet, but I'm closer (and suspect it's a Zwift issue).
>
> I had originally planned to installed the Wahoo app (Systm) to further troubleshoot, [...]

Sorry - Apple user in afraid. Samsung android tablets tend to also have ANT+?? = may be worth checking out (you will need the additional ant+ usb dongle and micro to standard USB adapter if your phone does have ant+)....

kschick, 2022-03-09:
I was struggling to get this to connect to Zwift on my android but it was connecting to Zwift on my macbook. I had only `Nearby devices` permission enabled for the Zwift app on android but I needed to enable  `Location` permission as well and that allowed me to connect.

carson-51, 2022-12-23:
I'm having the similar issue with the same setup. I tried similar troubleshooting steps.
One difference is that I was able to use it dozens of times initially and then, seemingly randomly during a ride, my cadence and power went to 0. After a reboot the next morning, it worked again for another 20+ rides. Then, it did a similar thing, except now, multiple reboots and a different SD card hasn't gotten it working again. I tried it on Rouvy, but that didn't work either. Calibrate works fine. I can see Flywheel bike on Bluetooth, but not Gymnasticon.
I'm trying to get a new Raspberry Pi but they're out of stock. Any ideas?

carson-51, 2022-12-27:
Update: I got it working with a different computer running Debian and a Bluetooth USB adapter (Zexmte). So I think it is my Raspberry Pi 4 that is bad.

4o4R, 2025-11-03:
I'm trying to update and fix and relearn to code (sorry for the issues as it's been like 20 years): https://github.com/4o4R/gymnasticonV2; I think I have addressed this issue.