ptx2/gymnasticon #95: IC4 disconnects immediately after connecting

particleandparcel, 2022-04-02:
Hi all,

I have been having a ton of fun getting set up with this great tool. I've been so frustrated by not being able to have indoor power data in Garmin or RideWithGPS, but my Edge 520 doesn't support a BLE power meter. Now I am finally going to be able to get those metrics all in one place!

However, I have only been able to get my bike to connect successfully twice, and it's unclear to me what conditions apply.

Bike: Schwinn IC4
Hardware: Mac Mini 2010 running Ubuntu 18.04, with Realtek Bluetooth USB device (Mac Mini onboard Bluetooth does not support BLE) and Garmin ANT+ stick
Software: Gymnasticon 1.4.1, node.js v16.14.2

Most of the time, as soon as I turn the bike on, the message I see in the log is some version of:

[2022-04-02T22:22:15.082Z] connecting to bike...
[2022-04-02T22:22:16.146Z] bike disconnected fc:74:30:8e:e1:46

I have been completely unable to figure out why, but sometimes (and not always) rebooting the computer seems to help. Restarting the gymnasticon service makes no difference. Down/up on the Bluetooth device doesn't help, nor does physically removing the Bluetooth device OR power-cycling the IC4.

I will be happy to post a log from btmon but I didn't want to do that completely unprompted. Please advise what you're looking for and I would be happy to help!

Thanks again. This is so awesome, and

particleandparcel, 2022-04-02:
Interestingly, I meant to say earlier but forgot that I also get a “Can’t open ANT+ stick” error, but I don’t know if that’s related to the above problems or if there’s a separate problem, like a lack of driver, perhaps. (This is using the Garmin ANT+ stick that came with my Forerunner watch.)

nealjane, 2022-04-04:
If your Garmin ant+ looks like this - it’s working fine on my pi zero/zero2
[image]

particleandparcel, 2022-04-05:
That is the same ANT+ adapter I have. Not sure what’s up with that. I might try upgrading the Ubuntu installation, see if a newer version of the kernel has a better Bluetooth driver. Setting aside the ANT issue, there has to be a reason that the IC4 only connects successfully once. 

Ubuntu 18.04 LTS uses the 4.15 kernel. Everything else is a fairly modern version, like Node.js.

If I could buy a Pi anywhere, I would, but I’m not seeing inventory at anywhere in person locally or online. That’s why I figured, I have a perfectly good old Mac Mini in use as a WiFi print server, might as well try that. 

Thanks.