ptx2/gymnasticon #96: Can't connect via ANT+ using version 1.4.1

rmaster78, 2022-04-29:
Just borrowed an ANT+ dongle from a friend and tried to see if Garmin Edge would pick up the ANT+ signal, but it isn't.  The bluetooth portion seems to be working as I see power/cadence in Zwift and cadence in Peloton app.

I typed "lsusb" in terminal window and it sees the ANT+ stick (see below)

pi@raspberrypi:~ $ lsusb
Bus 001 Device 002: ID 0fcf:1008 Dynastream Innovations, Inc. ANTUSB2 Stick
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Do I need to run some special command to get Gymnasticon to run or send ANT+ signal out so my Garmin head unit can see power?  Or is it just plug and play?

I did a search on Issues and Pull Requests and read all the related posts, but since I am not a SW programmer, I had hard time understanding it.

Thanks!

rmaster78, 2022-04-30:
Solved it by copy-pasting the entire command below to terminal (from pull request #36).  This created a file called "51-garmin-usb.rules" in the "/etc/udev/rules.d folder"

cat <<EOF |sudo tee /etc/udev/rules.d/51-garmin-usb.rules
# Garmin ANT+ - USB1
ATTRS{idVendor}=="0fcf", ATTRS{idProduct}=="1004", MODE="0666"
# Garmin ANT+ - USB2
ATTRS{idVendor}=="0fcf", ATTRS{idProduct}=="1008", MODE="0666"
# hLine USB2 ANT2
ATTRS{idVendor}=="0fcf", ATTRS{idProduct}=="1009", MODE="0666"
EOF

Once done, copy-paste the below command one by one

sudo udevadm control --reload-rules
sudo udevadm trigger

Spin the crank and tested with Garmin Edge 520 and it was able to find gymnasticon with ANT+ id 11234 and was displaying power and cadence.  

BUT... it seems like the ANT+ says ON all the time?  Garmin Edge 520 keeps connecting and disconnecting to power meter even tho I have Gymnasticon service stopped...

I read the issue where ANT+ staying ON all the time was resolved, but guess it isn't?? or did I just force things by typing in the above code and not implment it properly??

nealjane, 2022-04-30:
Unfortunately Ptx hasn’t been on GitHub in over a year, I believe @chriselsen helped/implemented the ant - so hopefully he may be able to help you.

chriselsen, 2022-04-30:
Not sure where you got build 1.4.1 from. The [last release listed here](https://github.com/ptx2/gymnasticon/releases) is 1.4.0, which is from Feb 21, 2021.
Pull request #80 fixes stopping the ANT+ server. But that got merged on Mar 28, 2021 and is therefore not in release 1.4.0. 

You could start with version 1.4.0 and the update to main and any subsequent pull request that you would like.

rmaster78, 2022-05-01:
Thanks for the responses guys.  Appreciate the time taken to answer my questions.

I ran the below command in termal to update my old version to the newest one

sudo npm install -g gymnasticon --unsafe-perm=true --allow-root

When I type "gymnasticon --version" It says 1.4.1

[image]

How do I do "update to main and any subsequent pull request" so the ANT+ stops when bike is not active?  Would you mind giving me some pointers or command lines to do this?  Sorry, I am not a programmer so I only have really basic knowledge of this.

Thank you so much!

rmaster78, 2022-05-03:
I tried the following commands in trying to install the branch that has the fix for ant+ issue, but results in same error... 

1) sudo npm install -g https://github.com/ptx2/gymnasticon.git#stop-ant-server-on-exit --unsafe-perm=true --allow-root
2) sudo npm install -g https://github.com/ptx2/gymnasticon.git#7a73123 --unsafe-perm=true --allow-root

I get the below error messages.

npm ERR! code ENOENT
npm ERR! syscall spawn git
npm ERR! path git
npm ERR! errno ENOENT
npm ERR! enoent Error while executing:
npm ERR! enoent undefined ls-remote -h -t https://github.com/ptx2/gymnasticon.git
npm ERR! enoent 
npm ERR! enoent 
npm ERR! enoent spawn git ENOENT
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent 

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2022-05-03T17_15_48_177Z-debug.log

Finally, I tried the below command

sudo npm install -g git@github.com:ptx2/gymnasticon.git#stop-ant-server-on-exit --unsafe-perm=true --allow-root

Got a different error message

npm ERR! code ENOLOCAL
npm ERR! Could not install from "github.com:ptx2/gymnasticon.git#stop-ant-server-on-exit" as it does not contain a package.json file.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2022-05-03T17_13_26_567Z-debug.log

Either my syntax is wrong or something is not setup on github for me to install this branch?

adanemayer, 2022-05-03:
@rmaster78 Here are the steps I ran to get the branch set up on my Raspberry Pi:

1. Setup internet access over Wifi
```
sudo apt-get update --allow-releaseinfo-change
sudo apt-get install git
mount -v | grep /boot
git clone https://github.com/ptx2/gymnasticon.git
cd gymnasticon
sudo service gymnasticon stop
sudo setcap cap_net_raw+eip $(eval readlink -f $(which node))
git fetch origin pull/89/head:antble-speed
git checkout antble-speed
npm install
npm run build
npm link
gymnasticon 
gymnasticon --version
```

rmaster78, 2022-05-05:
@adanemayer thanks!

after running the first command "sudo apt-get update --allow-releaseinfo-change" I got an error message stating "no space left on device".  Looks like I only have 1.6M of free space since I am using 2GB SD card lol.  I'll have to find/purchase new SD card and reinstall everything before trying again.

Question tho.. Since I am trying to install pull request #80 I guess the command for "git fetch" and "git checkout" should be as follows?

git fetch origin pull/80/head:stop-ant-server-on-exit
git checkout stop-ant-server-on-exit

Everything else seems like it can stay the same.

I'll report back in couple days once I get my new SD card.  Thanks for the help!

rmaster78, 2022-05-08:
Got the new SD card, installed the gymnasticon image and ran the commands listed above for pull request 80 and it is all working well!  Closing issue!

closed 2022-05-08 by rmaster78