ptx2/gymnasticon #97: Ride time not translating to Strava

adanemayer, 2022-05-03:
I've got a weird issue that I'm hoping is a simple fix. All my ride data streams into my Garmin Edge 25 without issue but when I upload to Strava they consistently show up as having been recorded 3 hours earlier (PST). I have confirmed that the system time is correct on both the Raspberry Pi and Garmin. I also tried recording rides with the Garmin time set manually. 

Without knowing where to look, my guess is that the gymnasticon service is somehow defaulting to PST. Please let me know what other info I can provide:

Schwinn IC4
Gymnasticon on branch `antble-speed`
Raspberry Pi 3B
Garmin ANT+ USB dongle
Garmin Edge 25

Sample log of a ride recorded this morning at 05:25 AM EST:
```
May  3 05:25:58 gymnasticon gymnasticon[32102]: [2022-05-03T09:25:58.716Z] wheel rotation [timestamp=1651569958715 revolutions=5649 speed=41.4km/h power=283W]
May  3 05:25:58 gymnasticon gymnasticon[32102]: [2022-05-03T09:25:58.720Z] pedal stroke [timestamp=1651569958720.9167 revolutions=1304 cadence=63rpm power=283W]
May  3 05:25:58 gymnasticon gymnasticon[32102]: [2022-05-03T09:25:58.897Z] wheel rotation [timestamp=1651569958897.261 revolutions=5650 speed=41.4km/h power=283W]
May  3 05:25:59 gymnasticon gymnasticon[32102]: [2022-05-03T09:25:59.015Z] received stats from bike [power=283W cadence=62rpm speed=41.4km/h]
May  3 05:25:59 gymnasticon gymnasticon[32102]: [2022-05-03T09:25:59.079Z] wheel rotation [timestamp=1651569959079.522 revolutions=5651 speed=41.4km/h power=283W]
May  3 05:25:59 gymnasticon gymnasticon[32102]: [2022-05-03T09:25:59.262Z] wheel rotation [timestamp=1651569959261.783 revolutions=5652 speed=41.4km/h power=283W]
May  3 05:25:59 gymnasticon gymnasticon[32102]: [2022-05-03T09:25:59.315Z] received stats from bike [power=290W cadence=61rpm speed=41.7km/h]
May  3 05:25:59 gymnasticon gymnasticon[32102]: [2022-05-03T09:25:59.442Z] wheel rotation [timestamp=1651569959442.7327 revolutions=5653 speed=41.7km/h power=290W]
May  3 05:25:59 gymnasticon gymnasticon[32102]: [2022-05-03T09:25:59.615Z] received stats from bike [power=283W cadence=63rpm speed=41.7km/h]
May  3 05:25:59 gymnasticon gymnasticon[32102]: [2022-05-03T09:25:59.623Z] wheel rotation [timestamp=1651569959623.6824 revolutions=5654 speed=41.7km/h power=283W]
May  3 05:25:59 gymnasticon gymnasticon[32102]: [2022-05-03T09:25:59.673Z] pedal stroke [timestamp=1651569959673.2976 revolutions=1305 cadence=63rpm power=283W]
May  3 05:25:59 gymnasticon gymnasticon[32102]: [2022-05-03T09:25:59.804Z] wheel rotation [timestamp=1651569959804.632 revolutions=5655 speed=41.7km/h power=283W]
May  3 05:25:59 gymnasticon gymnasticon[32102]: [2022-05-03T09:25:59.915Z] received stats from bike [power=269W cadence=62rpm speed=40.6km/h]
May  3 05:25:59 gymnasticon gymnasticon[32102]: [2022-05-03T09:25:59.990Z] wheel rotation [timestamp=1651569959990.4844 revolutions=5656 speed=40.6km/h power=269W]
```

rmaster78, 2022-05-03:
Not sure if you checked Strava (sorry if you did, but you did not mention that in your post), but it looks like if you don't have GPS data, it defaults to whatever timezone you set in Strava (see below link for FAQ)

https://support.strava.com/hc/en-us/articles/216515918-My-Activity-Has-the-Wrong-Date-or-Start-Time#:~:text=From%20the%20Strava%20website%2C%20open,zone%20is%20now%20set%20properly.

BTW, off-topic, but how did you install the branch?  I am trying to install a branch to stop ANT+ after the bike turns off, but don't know how to do this please read #96 if you can and help.  Thanks!

adanemayer, 2022-05-03:
Thank you @rmaster78 ! That  did it, can't believe I missed that setting.

closed 2022-05-03 by adanemayer