ptx2/gymnasticon #98: IC4 / Garmin wrong power output

romansemko, 2022-05-04:
So I have the gymnasticon running with ANT+ on a RPi Zero W, connecting to IC4.

IC4 has difficulty levels from 1-100. When I start pedalling at 32 difficulty, the reported power is about 140-160, which is fine. But when I start doing 3-minute intervals, where I switch the difficulty between 32 and 55, at some point gymnasticon reports pervious power as if I did not change the difficulty.

That is, if I pedal for 3 minutes at 32 the reported power is 140-160. Then I change the power to 55, but the reported power stays at 140-160 (should be 200-230). During half an hour of training, I need to restart the RPi once. Then the power is reported correctly again. This is not happening each time. On average, once every half an hour from what I have seen so far.

rmaster78, 2022-05-06:
Do you feel the resistance change in your legs when you encounter this issue?  I am sure you are as going from ~150W to ~220W is quite a bit.  

If you don't feel the resistance change in your legs, I am wondering if this has to do with the mechanical part of the dial or perhaps the IC4 bike itself is reporting incorrect power?  I think Gymnasticon works on just retransmitting the values spit out by the bike's computer and doesn't alter them unless you put the power-scale factors in the config file.  

I don't know how IC4 bike works, but I am guessing it is similar to the Flywheel spin bike I have.  On mine, the resistance is changed using a dial which is tied to a cable to change the resistance while riding.  Sometimes that cable gets stuck in the housing or doesn't release/pull all the way when I make small changes to the dial (like going from 40 to 42).  If this happens I just make larger changes on the dial until I see desired power on my Garmin head unit (It usually overshoots and I have to dial it back tho).  If you suspect that it is a mechanical issue, have you tried changing the resistance to a higher value to see if your issue persists?

romansemko, 2022-05-07:
The mechanical resistance is definitely there! I can feel it 😄 There is something like a red faucet that you spin in one direction or the other, which changes the difficulty (the distance of magnetic elements around the metallic wheel that make it change the resistance).

This is definitely not a problem in the bike for two reasons:

1. I tried connecting to Zwift on the Laptop, which reports the power flawlessly for the whole period of time.
2. Gymnasticon reports correct data as soon as I restart it while continuing spinning at the same power as before when the issue starts.

I also eliminated Garmin as the source of error by trying another Garmin device. My thought process was that maybe Gymnasticon reports correct data, but Garmin screws it up somehow and only starts reporting when re-connected back to Gymnasticon.

romansemko, 2022-05-07:
I can try pulling the whole log for the whole half an hour training if it helps somehow...

romansemko, 2022-05-07:
So I managed to log the time where the issue happened. I was doing my interval at about 56 difficulty level and then went down to difficulty 36, which should result the power dropping from 220 to 150-160.

However, my Garmin was still showing power about 220. Interestingly, in the logs the power drop is logged correctly from 220 to 150 and then to 110:

```
May  7 08:08:19 gymnasticon gymnasticon[318]: [2022-05-07T12:08:19.977Z] received stats from bike [power=207W cadence=57rpm speed=35.7km/h]
May  7 08:08:20 gymnasticon gymnasticon[318]: [2022-05-07T12:08:20.092Z] wheel rotation [timestamp=1651925300091.6318 revolutions=4418 speed=35.7km/h power=207W]
May  7 08:08:20 gymnasticon gymnasticon[318]: [2022-05-07T12:08:20.275Z] received stats from bike [power=207W cadence=57rpm speed=35.7km/h]
May  7 08:08:20 gymnasticon gymnasticon[318]: [2022-05-07T12:08:20.302Z] wheel rotation [timestamp=1651925300302.9932 revolutions=4419 speed=35.7km/h power=207W]
May  7 08:08:20 gymnasticon gymnasticon[318]: [2022-05-07T12:08:20.496Z] pedal stroke [timestamp=1651925300495.4578 revolutions=961 cadence=57rpm power=207W]
May  7 08:08:20 gymnasticon gymnasticon[318]: [2022-05-07T12:08:20.515Z] wheel rotation [timestamp=1651925300514.3545 revolutions=4420 speed=35.7km/h power=207W]
May  7 08:08:20 gymnasticon gymnasticon[318]: [2022-05-07T12:08:20.576Z] received stats from bike [power=207W cadence=57rpm speed=35.7km/h]
May  7 08:08:20 gymnasticon gymnasticon[318]: [2022-05-07T12:08:20.725Z] wheel rotation [timestamp=1651925300725.7158 revolutions=4421 speed=35.7km/h power=207W]
May  7 08:08:20 gymnasticon gymnasticon[318]: [2022-05-07T12:08:20.875Z] received stats from bike [power=207W cadence=57rpm speed=35.7km/h]
May  7 08:08:20 gymnasticon gymnasticon[318]: [2022-05-07T12:08:20.938Z] wheel rotation [timestamp=1651925300937.0771 revolutions=4422 speed=35.7km/h power=207W]
May  7 08:08:21 gymnasticon gymnasticon[318]: [2022-05-07T12:08:21.148Z] wheel rotation [timestamp=1651925301148.4385 revolutions=4423 speed=35.7km/h power=207W]
May  7 08:08:21 gymnasticon gymnasticon[318]: [2022-05-07T12:08:21.176Z] received stats from bike [power=195W cadence=56rpm speed=35.7km/h]
May  7 08:08:21 gymnasticon gymnasticon[318]: [2022-05-07T12:08:21.360Z] wheel rotation [timestamp=1651925301359.7998 revolutions=4424 speed=35.7km/h power=195W]
May  7 08:08:21 gymnasticon gymnasticon[318]: [2022-05-07T12:08:21.475Z] received stats from bike [power=195W cadence=57rpm speed=34.9km/h]
May  7 08:08:21 gymnasticon gymnasticon[318]: [2022-05-07T12:08:21.548Z] pedal stroke [timestamp=1651925301548.0894 revolutions=962 cadence=57rpm power=195W]
May  7 08:08:21 gymnasticon gymnasticon[318]: [2022-05-07T12:08:21.577Z] wheel rotation [timestamp=1651925301576.006 revolutions=4425 speed=34.9km/h power=195W]
May  7 08:08:21 gymnasticon gymnasticon[318]: [2022-05-07T12:08:21.775Z] received stats from bike [power=179W cadence=58rpm speed=33.8km/h]
May  7 08:08:21 gymnasticon gymnasticon[318]: [2022-05-07T12:08:21.800Z] wheel rotation [timestamp=1651925301799.2488 revolutions=4426 speed=33.8km/h power=179W]
May  7 08:08:22 gymnasticon gymnasticon[318]: [2022-05-07T12:08:22.023Z] wheel rotation [timestamp=1651925302022.4915 revolutions=4427 speed=33.8km/h power=179W]
May  7 08:08:22 gymnasticon gymnasticon[318]: [2022-05-07T12:08:22.075Z] received stats from bike [power=190W cadence=59rpm speed=34.5km/h]
May  7 08:08:22 gymnasticon gymnasticon[318]: [2022-05-07T12:08:22.242Z] wheel rotation [timestamp=1651925302241.2046 revolutions=4428 speed=34.5km/h power=190W]
May  7 08:08:22 gymnasticon gymnasticon[318]: [2022-05-07T12:08:22.375Z] received stats from bike [power=188W cadence=59rpm speed=34.5km/h]
May  7 08:08:22 gymnasticon gymnasticon[318]: [2022-05-07T12:08:22.459Z] wheel rotation [timestamp=1651925302459.9177 revolutions=4429 speed=34.5km/h power=188W]
May  7 08:08:22 gymnasticon gymnasticon[318]: [2022-05-07T12:08:22.566Z] pedal stroke [timestamp=1651925302565.0386 revolutions=963 cadence=59rpm power=188W]
May  7 08:08:22 gymnasticon gymnasticon[318]: [2022-05-07T12:08:22.675Z] received stats from bike [power=188W cadence=59rpm speed=34.5km/h]
May  7 08:08:22 gymnasticon gymnasticon[318]: [2022-05-07T12:08:22.687Z] wheel rotation [timestamp=1651925302680 revolutions=4430 speed=34.5km/h power=188W]
May  7 08:08:22 gymnasticon gymnasticon[318]: [2022-05-07T12:08:22.899Z] wheel rotation [timestamp=1651925302898.7131 revolutions=4431 speed=34.5km/h power=188W]
May  7 08:08:22 gymnasticon gymnasticon[318]: [2022-05-07T12:08:22.976Z] received stats from bike [power=182W cadence=58rpm speed=34km/h]
May  7 08:08:23 gymnasticon gymnasticon[318]: [2022-05-07T12:08:23.121Z] wheel rotation [timestamp=1651925303120.6426 revolutions=4432 speed=34km/h power=182W]
May  7 08:08:23 gymnasticon gymnasticon[318]: [2022-05-07T12:08:23.275Z] received stats from bike [power=170W cadence=59rpm speed=33.1km/h]
May  7 08:08:23 gymnasticon gymnasticon[318]: [2022-05-07T12:08:23.349Z] wheel rotation [timestamp=1651925303348.6064 revolutions=4433 speed=33.1km/h power=170W]
May  7 08:08:23 gymnasticon gymnasticon[318]: [2022-05-07T12:08:23.577Z] wheel rotation [timestamp=1651925303576.5703 revolutions=4434 speed=33.1km/h power=170W]
May  7 08:08:23 gymnasticon gymnasticon[318]: [2022-05-07T12:08:23.586Z] pedal stroke [timestamp=1651925303581.9878 revolutions=964 cadence=59rpm power=170W]
[... cut ...]
May  7 08:08:36 gymnasticon gymnasticon[318]: [2022-05-07T12:08:36.535Z] wheel rotation [timestamp=1651925316534.4587 revolutions=4488 speed=28km/h power=110W]
```

Not sure if that is a Garmin issue after all. Tried both Fenix 6 and 7 and both have shown this behaviour. Restarting RPi (hence, reconnecting Garmin) fixes the problem.

What I can test next time is re-connecting the Garmin sensors in the settings without restarting RPi and see if that fixes the issue. Then it would be an indication on Garmin side.

nealjane, 2022-05-08:
Don’t have an ic4, but do use gymnasticon with keiser and fenix 6 and never had anything like that,

Have you tried just writing another sd card with the software just in case it’s that simple an issue??

Gelma, 2022-05-18:
Sorry for the stupid question, but I use IC8 since one year (IC4/IC8 mostly the same with little difference in details like power cord and a few esthetic things).
I use it natively with Trainer Road, Zwift and Rouvy.  So, why do you use gymnasticon? (by the way, a very great project)

Asking just for curiosity to understand other pattern usage.

4o4R, 2025-11-03:
I'm trying to update and fix and relearn to code (sorry for the issues as it's been like 20 years): https://github.com/4o4R/gymnasticonV2; I think I have addressed this issue.