ptx2/gymnasticon #99: Adding additional ble dongle for 2 ble output

nealjane, 2023-07-30:
With apple watchOS10 beta including connecting to (ONLY ble) bike POWER/SPEED sensors. Users may want to connect to Apple Watch and Bluetooth app at same time - meaning they need x2 Bluetooth outputs from gymnasticon.

Is it possible to install a second Bluetooth 4 (ideally only 4.0 as they’re a lot cheaper than 4.1 or 5!) usb dongle to raspberry pi (zeroW/2zero/4) to provide x2 connections at same time??
@chriselsen

kard8, 2023-09-11:
Hi NealJane. Planning to use only power data to Apple watch (watchos 10) with no other app. I'm having an indoor cycle with BLE but it broadcasts as "Fitness machine" Profile. The apple watch recognises only Cycle Power / Cadence Sensor Profile. Will this project be helpful to implement the data?

nealjane, 2023-09-12:
Depends if your bike is on the list of compatible bikes 
- I was under the impression FTMS was supposed to work with Apple Watch too under iOS10??

kard8, 2023-09-12:
Thanks for the quick reply.

Unfortunately not yet, only Bluetooth bike accessories like cadence, speed
and power meters are detected in the beta version.

Wish Apple added support for them.

nealjane, 2023-09-29:
Not been any development on this software for quite some time unfortunately - ptx2 hasn’t been around for quite some time now (disappeared around covid - so have no idea what’s happened to him/her) Chris elsen did update with speed - but it’s not been copied into ptx2 this version..