ptx2/gymnasticon #100: Not able to install with npm

kard8, 2023-09-22:
The installation is failing after the npm command. Please help, I am new to the development scene.

Run on MacBook Air M2 with terminal in rosetta mode.
The latest xcode is installed.

Here is the npm output:

```
npm WARN deprecated node-pre-gyp@0.17.0: Please upgrade to @mapbox/node-pre-gyp: the non-scoped node-pre-gyp package is deprecated and only the @mapbox scoped package will recieve updates in the future
npm ERR! code 1
npm ERR! path /Users/(USER)/node_modules/gymnasticon/node_modules/usb
npm ERR! command failed
npm ERR! command sh -c prebuild-install --verbose || node-gyp rebuild
npm ERR! CC(target) Release/obj.target/libusb/libusb/libusb/core.o
npm ERR!   CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
npm ERR!   CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
npm ERR!   CC(target) Release/obj.target/libusb/libusb/libusb/io.o
npm ERR!   CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
npm ERR!   CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
npm ERR!   CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
npm ERR!   CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
npm ERR!   CC(target) Release/obj.target/libusb/libusb/libusb/os/darwin_usb.o
npm ERR!   LIBTOOL-STATIC Release/usb.a
npm ERR!   CXX(target) Release/obj.target/usb_bindings/src/node_usb.o
npm ERR! prebuild-install info begin Prebuild-install version 5.3.6
npm ERR! prebuild-install info install installing standalone, skipping download.
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@9.4.0
npm ERR! gyp info using node@18.18.0 | darwin | x64
npm ERR! gyp info find Python using Python version 3.11.5 found at "/Library/Frameworks/Python.framework/Versions/3.11/bin/python3"
npm ERR! gyp info spawn /Library/Frameworks/Python.framework/Versions/3.11/bin/python3
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
npm ERR! gyp info spawn args   'binding.gyp',
npm ERR! gyp info spawn args   '-f',
npm ERR! gyp info spawn args   'make',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/Users/(USER)/node_modules/gymnasticon/node_modules/usb/build/config.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/Users/(USER)/Library/Caches/node-gyp/18.18.0/include/node/common.gypi',
npm ERR! gyp info spawn args   '-Dlibrary=shared_library',
npm ERR! gyp info spawn args   '-Dvisibility=default',
npm ERR! gyp info spawn args   '-Dnode_root_dir=/Users/(USER)/Library/Caches/node-gyp/18.18.0',
npm ERR! gyp info spawn args   '-Dnode_gyp_dir=/usr/local/lib/node_modules/npm/node_modules/node-gyp',
npm ERR! gyp info spawn args   '-Dnode_lib_file=/Users/(USER)/Library/Caches/node-gyp/18.18.0/<(target_arch)/node.lib',
npm ERR! gyp info spawn args   '-Dmodule_root_dir=/Users/(USER)/node_modules/gymnasticon/node_modules/usb',
npm ERR! gyp info spawn args   '-Dnode_engine=v8',
npm ERR! gyp info spawn args   '--depth=.',
npm ERR! gyp info spawn args   '--no-parallel',
npm ERR! gyp info spawn args   '--generator-output',
npm ERR! gyp info spawn args   'build',
npm ERR! gyp info spawn args   '-Goutput_dir=.'
npm ERR! gyp info spawn args ]
npm ERR! gyp info spawn make
npm ERR! gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
npm ERR! ../libusb/libusb/core.c:2241:35: warning: comparison of integers of different signs: 'int' and 'unsigned long' [-Wsign-compare]
npm ERR!         if (header_len < 0 || header_len >= sizeof(buf)) {
npm ERR!                               ~~~~~~~~~~ ^  ~~~~~~~~~~~
npm ERR! ../libusb/libusb/core.c:2250:44: warning: comparison of integers of different signs: 'int' and 'unsigned long' [-Wsign-compare]
npm ERR!         if (text_len < 0 || text_len + header_len >= sizeof(buf)) {
npm ERR!                             ~~~~~~~~~~~~~~~~~~~~~ ^  ~~~~~~~~~~~
npm ERR! 2 warnings generated.
npm ERR! ../libusb/libusb/os/darwin_usb.c:344:3: warning: 'objc_registerThreadWithCollector' is deprecated: it does nothing. Define OBJC_SILENCE_GC_DEPRECATIONS=1 to temporarily silence this diagnostic. [-Wdeprecated-declarations]
npm ERR!   objc_registerThreadWithCollector();
npm ERR!   ^
npm ERR! /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/objc/objc-auto.h:248:1: note: 'objc_registerThreadWithCollector' has been explicitly marked deprecated here
npm ERR! OBJC_GC_DEPRECATED("it does nothing")
npm ERR! ^
[... 40 lines cut ...]
npm ERR!                                   ~~~~~~~~^
npm ERR! ../src/node_usb.cc:151:7: warning: 'MakeCallback' is deprecated [-Wdeprecated-declarations]
npm ERR!         Nan::MakeCallback(Nan::New(hotplugThis), "emit", 2, argv);
npm ERR!              ^
npm ERR! ../node_modules/nan/nan.h:1001:3: note: 'MakeCallback' has been explicitly marked deprecated here
npm ERR!   NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
npm ERR!   ^
npm ERR! ../node_modules/nan/nan.h:103:40: note: expanded from macro 'NAN_DEPRECATED'
npm ERR! # define NAN_DEPRECATED __attribute__((deprecated))
npm ERR!                                        ^
npm ERR! 2 warnings and 1 error generated.
npm ERR! make: *** [Release/obj.target/usb_bindings/src/node_usb.o] Error 1
npm ERR! gyp ERR! build error 
npm ERR! gyp ERR! stack Error: `make` failed with exit code: 2
npm ERR! gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:203:23)
npm ERR! gyp ERR! stack     at ChildProcess.emit (node:events:517:28)
npm ERR! gyp ERR! stack     at ChildProcess._handle.onexit (node:internal/child_process:292:12)
npm ERR! gyp ERR! System Darwin 22.6.0
npm ERR! gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd /Users/(USER)/node_modules/gymnasticon/node_modules/usb
npm ERR! gyp ERR! node -v v18.18.0
npm ERR! gyp ERR! node-gyp -v v9.4.0
npm ERR! gyp ERR! not ok

```

a-smiggle, 2024-06-23:
I have this same issue on a RPI running latest PI OS as of the 23 June 2024

a-smiggle, 2024-06-23:
I have tried Node 20 and 18.

kard8, 2024-07-12:
Found a great way by using the qdomyos-Zwift app. Works native on android and iOS. 

You can test your bike by using the nightly build on the repo.

The developer is kind to sort out any issue on your end with instant feedback.

cagnulein, 2024-07-12:
Thanks @kard8  :)