ptx2/gymnasticon #103: LifeFitness IC5 not connecting

bsjpark, 2025-03-01:
Hi all,
I have an old LifeFitness IC5, bought used. When I saw this project, this was my ticket in to make this smart indoor bike.
I got RasberryPi Zero W, as recommended. I don't seem to make it work still. I don't have Zwift but I wanted to use this for connecting to MyWhoosh. Any help, appreciated.

I've installed RasberryPi default OS package to ensure mine work. I've installed gymasticon. Tried to see if my bike shows up with power source, nothing showed up. I've looked it up and it seems I needed to calibrate. So, I went to FlyTest site (https://ptx2.net/apps/flytest/) and tried to calibrate.

I installed Chrome browser to enable BT. I've started by bike pedal rolling, got to FlyTest site got in. The "Connect" showed has shown on "Bike2", so I connected. From the FlyTest site, it shows pull resistance to 0, press next, to max, press next, and completed. When I looked back on my IC5 screen, it said connection failed. Not sure if this is a problem.

I continued on. Power cycled RasberryPi to boot Gymnasticon. I hooked up the monitor to see what's going on. Unfortunately, I don't have keyboard to hook up to mini-USB adapter nor usable keyboard to use.. I use Mac and my keyboards are BT or use Lightening cable.... sigh... In any rate, at least I get to see console outputs. It boots through, showing bunch of init sequences. It gets to prompt asking for password...? Odd, but this supposed to work as is. So, I turned by IC5 back on, started "Quick Start", nothing shows up nor on MyWoosh in connect to Power source, nothing showed up. I tried "Power Training" in IC5, still nothing shows up in MyWoosh.

My IC5 bike only connect through BT to my iPhone or iPad only to their own ICG App. Whenever, I use their app, it prompt to ask which bike it is connecting. Like "Bike 1" or "Bike 2". The Bike number seems to change. I am not sure if this is preventing from connect. When I look at BT list, my bike never shows up, only in ICG App. Not sure why and if this has something to do with.

Not sure what might be the issue.
- Am I doing the FlyTest calibration wrong? Please advice if I am and help me how I can do this properly. I feel stupid.
- Is there something different I need to do to have my IC5 bike connect to RasberryPi?

Any help appreciated. Thanks!

nealjane, 2025-03-01:
Hi, I’d suggest you just use the Gymnasticon image on the front page - just write the image to a blank sd card - it’s all setup and you’ll know that it’s working. 
The rest of your setup I can’t help with as I don’t have your bike model sorry.

bsjpark, 2025-03-03:
Hi nealjane,
Thanks for your input. I did install the Gymnasticon image on fresh SD-card. It booted as console but it doesn't seem to start properly. It's asking for login... and I don't have keyboard which can connect to micro USB. 😢

I wish I can attach the image here. I tried to copy/pasted from image... Anyone decipher what might be happening here?

thanks in advance..

1 Started Gymnasticon.
Started Daily apt download activities.
Started Daily apt upgrade and clean activities.
1 Reached target Timers.
Started triggerhappy global hotkey daemon,
Started System Logging Service,
Started Check for Raspberry Pi EEPROM updates.
Started rng-tools. service.
Started Save/Restore Sound Card State,
Started phys-swapfile - set up, mount/unmount, and delete
Started LSB: Switch to ondemand cpu governor (unless shift
Reached target Sound Card.
Started WPA supplicant.
1 Started Login Service.
| Started Avahi mDNS/DNS-SD Stack.
] Stopped Gymnasticon.
] Started Gymnasticon.
Stopped Gymnasticon.
] Started
Gymnasticon.
] Stopped Gymnasticon.
Started Gymnasticon.
e s 2 a/tcSats.ciod ny Um.
OR 1 Created slice systen-bthelper.slice.
Starting Bluetooth service...
OK 1 Started Bluetooth service.
1 Started Raspberry Pi bluetooth helper.
Starting Hostname Service...
OK 1 Started Hostname Service.
• 1 Stopped Gymnasticon.
OK 1 Started Gymnasticon.
O｝
1 Stopped Gymnasticon.
1 Started Gymnasticon.
1 Stopped Gymnasticon.
1 Started Gumnasticon.
Started dhopc on all interfaces.
Reached target Network.
Starting Permit User Sessions...
My
Starting /etc/rc. local Compatibility..
IP address is xxx.xxx.xxx.xxx fd39:1b10: f0a1:b147:df60:263b:1601 dde
1 Started /etc/rc, local Compatibility,
1 Started Permit User Sessions,
Started Getty on ttyl,
Reached
target Login Prompts.
Reached target Multi-User System.
1 Reached target Graphical Interface.
Starting Update UTMP about Systen Runlevel Changes..•
1 Started Update UTMP about System Runlevel Changes.
Raspbian GNU/Linux 10 raspberrypi ttyl
raspberrypi login:

nealjane, 2025-03-03:
Sorry - don't know what the issue is - writing this image should 'just work' - https://github.com/ptx2/gymnasticon/releases/latest/download/gymnasticon-raspberrypi.img.xz

bsjpark, 2025-03-04:
That's the image I programmed in the SD Card.
I got remote access enabled. I was able to do SSH into it.
Is there log or status I can pull to see?
The BT list didn't show my bike info there, so I'm kind of stuck.